AcutronicRobotics/moveit2 #4: CI multi-OS (enable OS X also)

vmayoral, 2019-02-20:
Motivated by #3 I'm considering whether it'll make sense to at least have CI also enabled for OS X.

I quickly checked and it seems it's feasible https://stackoverflow.com/questions/19818336/how-to-make-travis-ci-test-package-for-linux-os-x-windows.

vmayoral, 2019-02-24:
@LanderU, I saw https://github.com/AcutronicRobotics/moveit2/tree/ci-new, is this branch including OS X support? 

I just fixed master branch with some OS X-related errors. See https://github.com/AcutronicRobotics/moveit2/commit/c33b95cc53501ca4fb2c2e6bbfe243ee19935ec0

LanderU, 2019-02-24:
Not yet, I'm working to have minimal version on Linux working first. I'm reviewing/porting all the stuff. When I finish with this I'll include the OS X support as well.

vmayoral, 2019-02-24:
Cool, many thanks.

LanderU, 2019-02-27:
CI enabled for OS X on master.

closed 2019-02-27 by LanderU