AcutronicRobotics/moveit2 #13: robot_model unit tests failing

vmayoral, 2019-02-26:
Failing when enabling units tests of robot_model submodule of [moveit_core](https://github.com/AcutronicRobotics/moveit2/blob/master/moveit_core/robot_model/CMakeLists.txt#L25-L32):

```bash
Undefined symbols for architecture x86_64:
  "random_numbers::RandomNumberGenerator::quaternion(double*)", referenced from:
      moveit::core::FloatingJointModel::getVariableRandomPositions(random_numbers::RandomNumberGenerator&, double*, std::__1::vector<moveit::core::VariableBounds, std::__1::allocator<moveit::core::VariableBounds> > const&) const in libmoveit_robot_model.a(floating_joint_model.cpp.o)
      moveit::core::FloatingJointModel::getVariableRandomPositionsNearBy(random_numbers::RandomNumberGenerator&, double*, std::__1::vector<moveit::core::VariableBounds, std::__1::allocator<moveit::core::VariableBounds> > const&, double const*, double) const in libmoveit_robot_model.a(floating_joint_model.cpp.o)
  "srdf::SRDFWriter::updateSRDFModel(urdf::ModelInterface const&)", referenced from:
      moveit::core::RobotModelBuilder::build() in libmoveit_test_utils.a(robot_model_test_utils.cpp.o)
  "srdf::SRDFWriter::SRDFWriter()", referenced from:
      moveit::core::RobotModelBuilder::RobotModelBuilder(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) in libmoveit_test_utils.a(robot_model_test_utils.cpp.o)
  "srdf::SRDFWriter::~SRDFWriter()", referenced from:
      std::__1::shared_ptr<srdf::SRDFWriter>::shared_ptr<srdf::SRDFWriter>(srdf::SRDFWriter*, std::__1::enable_if<is_convertible<srdf::SRDFWriter*, srdf::SRDFWriter*>::value, std::__1::shared_ptr<srdf::SRDFWriter>::__nat>::type) in libmoveit_test_utils.a(robot_model_test_utils.cpp.o)
      std::__1::__shared_ptr_pointer<srdf::SRDFWriter*, std::__1::default_delete<srdf::SRDFWriter>, std::__1::allocator<srdf::SRDFWriter> >::__on_zero_shared() in libmoveit_test_utils.a(robot_model_test_utils.cpp.o)
  "srdf::Model::initFile(urdf::ModelInterface const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)", referenced from:
      moveit::core::loadSRDFModel(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) in libmoveit_test_utils.a(robot_model_test_utils.cpp.o)
  "boost::filesystem::path::operator/=(boost::filesystem::path const&)", referenced from:
      boost::filesystem::operator/(boost::filesystem::path const&, boost::filesystem::path const&) in libmoveit_test_utils.a(robot_model_test_utils.cpp.o)
  "boost::basic_regex<char, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::do_assign(char const*, char const*, unsigned int)", referenced from:
      boost::basic_regex<char, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::assign(char const*, char const*, unsigned int) in libmoveit_test_utils.a(robot_model_test_utils.cpp.o)
  "boost::re_detail_106800::get_mem_block()", referenced from:
      boost::re_detail_106800::perl_matcher<std::__1::__wrap_iter<char const*>, std::__1::allocator<boost::sub_match<std::__1::__wrap_iter<char const*> > >, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::extend_stack() in libmoveit_test_utils.a(robot_model_test_utils.cpp.o)
      boost::re_detail_106800::save_state_init::save_state_init(boost::re_detail_106800::saved_state**, boost::re_detail_106800::saved_state**) in libmoveit_test_utils.a(robot_model_test_utils.cpp.o)
  "boost::re_detail_106800::put_mem_block(void*)", referenced from:
      boost::re_detail_106800::perl_matcher<std::__1::__wrap_iter<char const*>, std::__1::allocator<boost::sub_match<std::__1::__wrap_iter<char const*> > >, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::unwind_extra_block(bool) in libmoveit_test_utils.a(robot_model_test_utils.cpp.o)
      boost::re_detail_106800::save_state_init::~save_state_init() in libmoveit_test_utils.a(robot_model_test_utils.cpp.o)
  "boost::re_detail_106800::verify_options(unsigned int, boost::regex_constants::_match_flags)", referenced from:
      boost::re_detail_106800::perl_matcher<std::__1::__wrap_iter<char const*>, std::__1::allocator<boost::sub_match<std::__1::__wrap_iter<char const*> > >, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::find_imp() in libmoveit_test_utils.a(robot_model_test_utils.cpp.o)
  "boost::re_detail_106800::raise_runtime_error(std::runtime_error const&)", referenced from:
      void boost::re_detail_106800::raise_error<boost::regex_traits_wrapper<boost::regex_traits<char, boost::cpp_regex_traits<char> > > >(boost::regex_traits_wrapper<boost::regex_traits<char, boost::cpp_regex_traits<char> > > const&, boost::regex_constants::error_type) in libmoveit_test_utils.a(robot_model_test_utils.cpp.o)
  "boost::re_detail_106800::get_default_error_string(boost::regex_constants::error_type)", referenced from:
      boost::re_detail_106800::cpp_regex_traits_implementation<char>::error_string(boost::regex_constants::error_type) const in libmoveit_test_utils.a(robot_model_test_utils.cpp.o)
  "boost::system::detail::generic_category_instance", referenced from:
      boost::system::generic_category() in test.cpp.o
      boost::system::generic_category() in libmoveit_test_utils.a(robot_model_test_utils.cpp.o)
      boost::system::generic_category() in libmoveit_robot_model.a(robot_model.cpp.o)
  "moveit::tools::Profiler::end(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)", referenced from:
      moveit::tools::Profiler::ScopedBlock::~ScopedBlock() in libmoveit_robot_model.a(robot_model.cpp.o)
  "moveit::tools::Profiler::stop()", referenced from:
      moveit::tools::Profiler::ScopedStart::~ScopedStart() in libmoveit_robot_model.a(robot_model.cpp.o)
  "moveit::tools::Profiler::begin(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)", referenced from:
      moveit::tools::Profiler::ScopedBlock::ScopedBlock(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, moveit::tools::Profiler&) in libmoveit_robot_model.a(robot_model.cpp.o)
  "moveit::tools::Profiler::start()", referenced from:
      moveit::tools::Profiler::ScopedStart::ScopedStart(moveit::tools::Profiler&) in libmoveit_robot_model.a(robot_model.cpp.o)
  "moveit::tools::Profiler::status(std::__1::basic_ostream<char, std::__1::char_traits<char> >&, bool)", referenced from:
      moveit::tools::Profiler::Status(std::__1::basic_ostream<char, std::__1::char_traits<char> >&, bool) in test.cpp.o
  "moveit::tools::Profiler::instance()", referenced from:
      moveit::tools::Profiler::Status(std::__1::basic_ostream<char, std::__1::char_traits<char> >&, bool) in test.cpp.o
      moveit::core::RobotModel::buildModel(urdf::ModelInterface const&, srdf::Model const&) in libmoveit_robot_model.a(robot_model.cpp.o)
      moveit::core::RobotModel::buildJointInfo() in libmoveit_robot_model.a(robot_model.cpp.o)
      moveit::core::RobotModel::constructShape(urdf::Geometry const*) in libmoveit_robot_model.a(robot_model.cpp.o)
  "moveit::Exception::Exception(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)", referenced from:
      moveit::core::RobotModel::getVariableIndex(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) const in libmoveit_robot_model.a(robot_model.cpp.o)
      moveit::core::JointModel::getLocalVariableIndex(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) const in libmoveit_robot_model.a(joint_model.cpp.o)
      moveit::core::JointModelGroup::getVariableRandomPositionsNearBy(random_numbers::RandomNumberGenerator&, double*, std::__1::vector<std::__1::vector<moveit::core::VariableBounds, std::__1::allocator<moveit::core::VariableBounds> > const*, std::__1::allocator<std::__1::vector<moveit::core::VariableBounds, std::__1::allocator<moveit::core::VariableBounds> > const*> > const&, double const*, std::__1::vector<double, std::__1::allocator<double> > const&) const in libmoveit_robot_model.a(joint_model_group.cpp.o)
  "shapes::computeShapeExtents(shapes::Shape const*)", referenced from:
      moveit::core::LinkModel::setGeometry(std::__1::vector<std::__1::shared_ptr<shapes::Shape const>, std::__1::allocator<std::__1::shared_ptr<shapes::Shape const> > > const&, std::__1::vector<Eigen::Transform<double, 3, 1, 0> [... cut ...] n::Transform<double, 3, 1, 0> > > const&) in libmoveit_robot_model.a(link_model.cpp.o)
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [robot_model/test_robot_model] Error 1
make[1]: *** [robot_model/CMakeFiles/test_robot_model.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2
---
Failed   <<< moveit_core	[ Exited with code 2 ]

Summary: 0 packages finished [27.3s]
```

davetcoleman, 2019-03-06:
@mlautman could you help debug this failing test?

vmayoral, 2019-03-10:
I've been looking at the unit tests of other submodules and I've been getting somewhat similar errors so I may have something wrong in my local setup (working with OS X). 

I re-enabled unit tests for this module in a different branch, results should come up at https://travis-ci.org/AcutronicRobotics/moveit2/builds/504368925?utm_source=github_status&utm_medium=notification

@mlautman, any chance you managed to shed some light into this?

mlautman, 2019-03-15:
Looking into it

vmayoral, 2019-03-24:
Minor improvement after [reviewing dependencies](https://github.com/AcutronicRobotics/moveit2/commit/caa10a04305e11cf83322ddd9196f04511542863) but still obtaining an error:
```bash
Undefined symbols for architecture x86_64:
  "boost::system::detail::generic_category_instance", referenced from:
      boost::system::generic_category() in test.cpp.o
  "moveit::core::RobotModelBuilder::addVirtualJoint(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)", referenced from:
      SiblingAssociateLinks_SimpleYRobot_Test::TestBody() in test.cpp.o
  "moveit::core::RobotModelBuilder::build()", referenced from:
      SiblingAssociateLinks_SimpleYRobot_Test::TestBody() in test.cpp.o
  "moveit::core::RobotModelBuilder::isValid()", referenced from:
      SiblingAssociateLinks_SimpleYRobot_Test::TestBody() in test.cpp.o
  "moveit::core::RobotModelBuilder::addChain(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::vector<geometry_msgs::msg::Pose_<std::__1::allocator<void> >, std::__1::allocator<geometry_msgs::msg::Pose_<std::__1::allocator<void> > > > const&)", referenced from:
      SiblingAssociateLinks_SimpleYRobot_Test::TestBody() in test.cpp.o
  "moveit::core::RobotModelBuilder::addGroup(std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > const&, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)", referenced from:
      SiblingAssociateLinks_SimpleYRobot_Test::TestBody() in test.cpp.o
  "moveit::core::RobotModelBuilder::RobotModelBuilder(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)", referenced from:
      SiblingAssociateLinks_SimpleYRobot_Test::TestBody() in test.cpp.o
  "moveit::core::loadTestingRobotModel(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)", referenced from:
      LoadPlanningModelsPr2::SetUp() in test.cpp.o
  "moveit::tools::Profiler::status(std::__1::basic_ostream<char, std::__1::char_traits<char> >&, bool)", referenced from:
      moveit::tools::Profiler::Status(std::__1::basic_ostream<char, std::__1::char_traits<char> >&, bool) in test.cpp.o
  "moveit::tools::Profiler::instance()", referenced from:
      moveit::tools::Profiler::Status(std::__1::basic_ostream<char, std::__1::char_traits<char> >&, bool) in test.cpp.o
  "rclcpp::get_logger(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)", referenced from:
      ___cxx_global_var_init in test.cpp.o
  "moveit::core::RobotModel::getLinkModel(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) const", referenced from:
      SiblingAssociateLinks_SimpleYRobot_Test::TestBody() in test.cpp.o
  "moveit::core::RobotModel::getJointModel(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) const", referenced from:
      LoadPlanningModelsPr2_Model_Test::TestBody() in test.cpp.o
ld: symbol(s) not found for architecture x86_64
```

vmayoral, 2019-03-24:
Something similar happens with the unit tests from other modules (e.g. `robot_state` submodule of `moveit_core`).

@dirk-thomas, @wjwwood it seems we're blocked here, any chance you can have a look at https://github.com/AcutronicRobotics/moveit2/blob/master/moveit_core/robot_model/CMakeLists.txt#L26 and advice whether there's something we're missing (refer to the previous comment for the error we're getting)?

davetcoleman, 2019-03-25:
I think @mlautman 's recent moveit1 fix will help this:
https://github.com/ros-planning/moveit/pull/1409

dirk-thomas, 2019-03-25:
According to the error message you are missing some libraries to provide the missing symbols. Looking at the commented CMake code you seem to pass wrong arguments to `ament_target_dependencies`. Its pdocblock](https://github.com/ament/ament_cmake/blob/122f283ab156c159194afce3d932f96327d28ff1/ament_cmake_target_dependencies/cmake/ament_target_dependencies.cmake#L15-L30) states that it takes the target name as well as a list of package names as arguments - nothing else. So passing `*_LIBRARIES` is wrong.

vmayoral, 2019-03-30:
Thanks @dirk-thomas, https://github.com/AcutronicRobotics/moveit2/commit/e0dca76edbf906a56d6d0a492f07a124dc79c03e addressed some of the linking issues but still failing:

```bash
Undefined symbols for architecture x86_64:
  "moveit::core::RobotModelBuilder::addVirtualJoint(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)", referenced from:
      SiblingAssociateLinks_SimpleYRobot_Test::TestBody() in test.cpp.o
  "moveit::core::RobotModelBuilder::build()", referenced from:
      SiblingAssociateLinks_SimpleYRobot_Test::TestBody() in test.cpp.o
  "moveit::core::RobotModelBuilder::isValid()", referenced from:
      SiblingAssociateLinks_SimpleYRobot_Test::TestBody() in test.cpp.o
  "moveit::core::RobotModelBuilder::addChain(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::vector<geometry_msgs::msg::Pose_<std::__1::allocator<void> >, std::__1::allocator<geometry_msgs::msg::Pose_<std::__1::allocator<void> > > > const&)", referenced from:
      SiblingAssociateLinks_SimpleYRobot_Test::TestBody() in test.cpp.o
  "moveit::core::RobotModelBuilder::addGroup(std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > const&, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)", referenced from:
      SiblingAssociateLinks_SimpleYRobot_Test::TestBody() in test.cpp.o
  "moveit::core::RobotModelBuilder::RobotModelBuilder(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)", referenced from:
      SiblingAssociateLinks_SimpleYRobot_Test::TestBody() in test.cpp.o
  "moveit::core::loadTestingRobotModel(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)", referenced from:
      LoadPlanningModelsPr2::SetUp() in test.cpp.o
  "moveit::tools::Profiler::status(std::__1::basic_ostream<char, std::__1::char_traits<char> >&, bool)", referenced from:
      moveit::tools::Profiler::Status(std::__1::basic_ostream<char, std::__1::char_traits<char> >&, bool) in test.cpp.o
  "moveit::tools::Profiler::instance()", referenced from:
      moveit::tools::Profiler::Status(std::__1::basic_ostream<char, std::__1::char_traits<char> >&, bool) in test.cpp.o
  "moveit::core::RobotModel::getLinkModel(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) const", referenced from:
      SiblingAssociateLinks_SimpleYRobot_Test::TestBody() in test.cpp.o
  "moveit::core::RobotModel::getJointModel(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) const", referenced from:
      LoadPlanningModelsPr2_Model_Test::TestBody() in test.cpp.o
ld: symbol(s) not found for architecture x86_64
```

@davetcoleman and @mlautman, I tried fetching the changes from @mlautman patch at https://github.com/AcutronicRobotics/moveit2/commit/e0dca76edbf906a56d6d0a492f07a124dc79c03e but the result is the one above. Reviewing the linking issues, most point to `moveit_robot_model` and `moveit_test_utils`. 

Is there something that I missed? Help here is required.

vmayoral, 2019-03-30:
Re-tried again @mlautman's changes more directly https://github.com/AcutronicRobotics/moveit2/commit/e841cafa06443a36e364ee2954eb22efa3af241f

Similar result: 
```bash
 "moveit::core::RobotModelBuilder::addVirtualJoint(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)", referenced from:
      SiblingAssociateLinks_SimpleYRobot_Test::TestBody() in test.cpp.o
  "moveit::core::RobotModelBuilder::build()", referenced from:
      SiblingAssociateLinks_SimpleYRobot_Test::TestBody() in test.cpp.o
  "moveit::core::RobotModelBuilder::isValid()", referenced from:
      SiblingAssociateLinks_SimpleYRobot_Test::TestBody() in test.cpp.o
  "moveit::core::RobotModelBuilder::addChain(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::vector<geometry_msgs::msg::Pose_<std::__1::allocator<void> >, std::__1::allocator<geometry_msgs::msg::Pose_<std::__1::allocator<void> > > > const&)", referenced from:
      SiblingAssociateLinks_SimpleYRobot_Test::TestBody() in test.cpp.o
  "moveit::core::RobotModelBuilder::addGroup(std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > const&, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)", referenced from:
      SiblingAssociateLinks_SimpleYRobot_Test::TestBody() in test.cpp.o
  "moveit::core::RobotModelBuilder::RobotModelBuilder(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)", referenced from:
      SiblingAssociateLinks_SimpleYRobot_Test::TestBody() in test.cpp.o
  "moveit::core::loadTestingRobotModel(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)", referenced from:
      LoadPlanningModelsPr2::SetUp() in test.cpp.o
  "moveit::tools::Profiler::status(std::__1::basic_ostream<char, std::__1::char_traits<char> >&, bool)", referenced from:
      moveit::tools::Profiler::Status(std::__1::basic_ostream<char, std::__1::char_traits<char> >&, bool) in test.cpp.o
  "moveit::tools::Profiler::instance()", referenced from:
      moveit::tools::Profiler::Status(std::__1::basic_ostream<char, std::__1::char_traits<char> >&, bool) in test.cpp.o
  "moveit::core::RobotModel::getLinkModel(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) const", referenced from:
      SiblingAssociateLinks_SimpleYRobot_Test::TestBody() in test.cpp.o
  "moveit::core::RobotModel::getJointModel(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) const", referenced from:
      LoadPlanningModelsPr2_Model_Test::TestBody() in test.cpp.o
ld: symbol(s) not found for architecture x86_64
```

There's some weird issue with `moveit_test_utils`.

vmayoral, 2019-03-31:
Connected to https://github.com/ros-planning/moveit2/pull/68, still unable to get the right symbols.

vmayoral, 2019-04-01:
ping @mlautman and @davetcoleman, can you please help with this?

mlautman, 2019-04-03:
@vmayoral Can you send over some instructions to reproduce your setup so that we can help you debug this?

vmayoral, 2019-04-04:
@mlautman https://acutronicrobotics.com/docs/products/robots/mara/moveit2/install/ubuntu should do for getting the environment set up. Then simply uncomment https://github.com/AcutronicRobotics/moveit2/blob/master/moveit_core/robot_model/CMakeLists.txt#L31-L52 and `colcon build --packages-select moveit_core`. Let me know if that doesn't work and you have issues reproducing the error.

vmayoral, 2019-04-06:
Already pinged the PickNik guys on this (@mlautman and @davetcoleman) but we together, seem to be in a stalemate. 

@dirk-thomas and @wjwwood, sorry to bother, any chance you have some input about `moveit_test_utils` as we're currently porting it https://github.com/AcutronicRobotics/moveit2/blob/master/moveit_core/utils/CMakeLists.txt?

vmayoral, 2019-04-08:
Updated the title, this happens across packages in moveit2 codebase. Need further assistance to continue porting tests.

dirk-thomas, 2019-04-08:
I was unable to simply clone and build this repo in a workspace successfully. Therefore I can only describe what might be causing your problem.

* the test failing to link is in `moveit_core/robot_model/CMakeLists.txt`
* the symbol `moveit::core::RobotModelBuilder::addVirtualJoint` isn't found during linking
* the symbol is defined in `moveit_core/utils/src/robot_model_test_utils.cpp`
* which is part of the library `moveit_test_utils`
* which is added in `moveit_core/utils/CMakeLists.txt`

But based on the order of added subdirectories `utils` is added *after* `robot_model` in `moveit_core/CMakeLists.txt` so the library containing the symbol hasn't even been defined yet when you try to use it.

vmayoral, 2019-04-08:
Thanks @dirk-thomas for the input, let me process all of that and retry it once again. In case I fail, I'll make sure to provide an environment that easily reproduces our current problem.

vmayoral, 2019-04-08:
Note to self: take into account https://travis-ci.org/AcutronicRobotics/moveit2/builds/517131210#L1859

vmayoral, 2019-04-14:
> But based on the order of added subdirectories `utils` is added _after_ `robot_model` in `moveit_core/CMakeLists.txt` so the library containing the symbol hasn't even been defined yet when you try to use it.

Changed order  of added subdirectories and being tested at https://travis-ci.org/AcutronicRobotics/moveit2/builds/519855880?utm_source=github_status&utm_medium=notification

vmayoral, 2019-04-14:
Updated comment above, still failing. @dirk-thomas I've prepared a branch that allows to reproduce the issue locally through a docker container. Find below the instructions:

```bash
git clone https://github.com/acutronicrobotics/moveit2 -b robot-model-tests4
cd moveit2
docker build -t moveit2-robot-model-test .docker/ci/
docker run -it moveit2-robot-model-test
```

making sure the Docker VMs have enough memory result into something like:

```bash
CMakeFiles/test_robot_model.dir/test/test.cpp.o: In function `LoadPlanningModelsPr2_Model_Test::TestBody()':
test.cpp:(.text+0x4b3): undefined reference to `moveit::core::RobotModel::getJointModel(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
CMakeFiles/test_robot_model.dir/test/test.cpp.o: In function `SiblingAssociateLinks_SimpleYRobot_Test::TestBody()':
test.cpp:(.text+0x83a): undefined reference to `moveit::core::RobotModelBuilder::RobotModelBuilder(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
test.cpp:(.text+0x901): undefined reference to `moveit::core::RobotModelBuilder::addChain(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::vector<geometry_msgs::msg::Pose_<std::allocator<void> >, std::allocator<geometry_msgs::msg::Pose_<std::allocator<void> > > > const&)'
test.cpp:(.text+0x9d7): undefined reference to `moveit::core::RobotModelBuilder::addChain(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::vector<geometry_msgs::msg::Pose_<std::allocator<void> >, std::allocator<geometry_msgs::msg::Pose_<std::allocator<void> > > > const&)'
test.cpp:(.text+0xaad): undefined reference to `moveit::core::RobotModelBuilder::addChain(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::vector<geometry_msgs::msg::Pose_<std::allocator<void> >, std::allocator<geometry_msgs::msg::Pose_<std::allocator<void> > > > const&)'
test.cpp:(.text+0xb83): undefined reference to `moveit::core::RobotModelBuilder::addChain(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::vector<geometry_msgs::msg::Pose_<std::allocator<void> >, std::allocator<geometry_msgs::msg::Pose_<std::allocator<void> > > > const&)'
test.cpp:(.text+0xcac): undefined reference to `moveit::core::RobotModelBuilder::addVirtualJoint(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
test.cpp:(.text+0xdf9): undefined reference to `moveit::core::RobotModelBuilder::addGroup(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
test.cpp:(.text+0xe87): undefined reference to `moveit::core::RobotModelBuilder::isValid()'
test.cpp:(.text+0xfbb): undefined reference to `moveit::core::RobotModelBuilder::build()'
test.cpp:(.text+0x12bf): undefined reference to `moveit::core::RobotModel::getLinkModel(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
CMakeFiles/test_robot_model.dir/test/test.cpp.o: In function `moveit::tools::Profiler::Status(std::ostream&, bool)':
test.cpp:(.text._ZN6moveit5tools8Profiler6StatusERSob[_ZN6moveit5tools8Profiler6StatusERSob]+0x12): undefined reference to `moveit::tools::Profiler::instance()'
test.cpp:(.text._ZN6moveit5tools8Profiler6StatusERSob[_ZN6moveit5tools8Profiler6StatusERSob]+0x28): undefined reference to `moveit::tools::Profiler::status(std::ostream&, bool)'
CMakeFiles/test_robot_model.dir/test/test.cpp.o: In function `LoadPlanningModelsPr2::SetUp()':
test.cpp:(.text._ZN21LoadPlanningModelsPr25SetUpEv[_ZN21LoadPlanningModelsPr25SetUpEv]+0x4e): undefined reference to `moveit::core::loadTestingRobotModel(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
```

vmayoral, 2019-04-14:
I didn't investigate it much but on a quick look, it seems that I can see the missing symbols in the local `libmoveit_test_utils` library when inspecting manually.

```bash
| victor:~/ros2_moveit_ws/build/moveit_core/utils$
| => nm libmoveit_test_utils.a | grep addVirtualJoint
000000000000ba80 T __ZN6moveit4core17RobotModelBuilder15addVirtualJointERKNSt3__112basic_stringIcNS2_11char_traitsIcEENS2_9allocatorIcEEEESA_SA_SA_
```

mlautman, 2019-04-16:
I am having trouble reproducing your results. The way that you have set up the docker-entrypoint.sh makes it really hard to iterate without rebuilding the entire workspace from scratch each time (approx 10 min)

Looking at that error, the problem might be that `moveit_test_utils` only gets built if you are building the tests. (https://github.com/AcutronicRobotics/moveit2/blob/1bc57d0aabf23deedca075f38f86356a87dfeb29/moveit_core/utils/CMakeLists.txt#L19) What happens if you build that library outside of the `if(BUILD_TESTING)` block?

mlautman, 2019-04-16:
You may want to look into using an overlay to avoid building dependencies each time

mlautman, 2019-04-16:
I have been trying to debug the issue but there seem to be quite a few issues with that branch which make the build fail. Namely including header files from srdfdom and random_numbers.h are both failing.  I also tried building master but it has the same issue.

anasarrak, 2019-04-16:
@mlautman Taking into account that you only want to compile moveit_core, just by using ```colcon build --merge-install --packages-select moveit_core``` should work.

> Namely including header files from srdfdom and random_numbers.h are both failing. I also tried building master but it has the same issue.

Which kind of errors?

mlautman, 2019-04-16:
@anasarrak, The build command is in the docker_entrypoint.sh file so anytime you do `docker run -it ....` it builds the entire project. This could be avoided by having an underlay that is built in the Dockerfile but this is not the most pressing issue

mlautman, 2019-04-16:
This is the error I get:
```
--- stderr: moveit_core                             
In file included from /home/mike/ws_acutronics/src/moveit2/moveit_core/robot_model/include/moveit/robot_model/fixed_joint_model.h:40:0,
                 from /home/mike/ws_acutronics/src/moveit2/moveit_core/robot_model/src/fixed_joint_model.cpp:37:
/home/mike/ws_acutronics/src/moveit2/moveit_core/robot_model/include/moveit/robot_model/joint_model.h:46:10: fatal error: random_numbers/random_numbers.h: No such file or directory
 #include <random_numbers/random_numbers.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [robot_model/CMakeFiles/moveit_robot_model.dir/src/fixed_joint_model.cpp.o] Error 1
make[1]: *** [robot_model/CMakeFiles/moveit_robot_model.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
In file included from /home/mike/ws_acutronics/src/moveit2/moveit_core/kinematic_constraints/include/moveit/kinematic_constraints/kinematic_constraint.h:40:0,
                 from /home/mike/ws_acutronics/src/moveit2/moveit_core/kinematic_constraints/src/kinematic_constraint.cpp:37:
/home/mike/ws_acutronics/src/moveit2/moveit_core/robot_model/include/moveit/robot_model/robot_model.h:43:10: fatal error: srdfdom/model.h: No such file or directory
 #include <srdfdom/model.h>
          ^~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [kinematic_constraints/CMakeFiles/moveit_kinematic_constraints.dir/src/kinematic_constraint.cpp.o] Error 1
make[1]: *** [kinematic_constraints/CMakeFiles/moveit_kinematic_constraints.dir/all] Error 2
In file included from /home/mike/ws_acutronics/src/moveit2/moveit_core/robot_model/include/moveit/robot_model/joint_model_group.h:41:0,
                 from /home/mike/ws_acutronics/src/moveit2/moveit_core/kinematics_base/src/kinematics_base.cpp:38:
/home/mike/ws_acutronics/src/moveit2/moveit_core/robot_model/include/moveit/robot_model/joint_model.h:46:10: fatal error: random_numbers/random_numbers.h: No such file or directory
 #include <random_numbers/random_numbers.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [kinematics_base/CMakeFiles/moveit_kinematics_base.dir/src/kinematics_base.cpp.o] Error 1
make[1]: *** [kinematics_base/CMakeFiles/moveit_kinematics_base.dir/all] Error 2
In file included from /home/mike/ws_acutronics/src/moveit2/moveit_core/collision_detection/include/moveit/collision_detection/collision_common.h:47:0,
                 from /home/mike/ws_acutronics/src/moveit2/moveit_core/collision_detection/include/moveit/collision_detection/collision_matrix.h:40,
                 from /home/mike/ws_acutronics/src/moveit2/moveit_core/collision_detection/include/moveit/collision_detection/collision_robot.h:40,
                 from /home/mike/ws_acutronics/src/moveit2/moveit_core/collision_detection/include/moveit/collision_detection/allvalid/collision_robot_allvalid.h:40,
                 from /home/mike/ws_acutronics/src/moveit2/moveit_core/collision_detection/src/allvalid/collision_robot_allvalid.cpp:37:
/home/mike/ws_acutronics/src/moveit2/moveit_core/robot_model/include/moveit/robot_model/robot_model.h:43:10: fatal error: srdfdom/model.h: No such file or directory
 #include <srdfdom/model.h>
          ^~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [collision_detection/CMakeFiles/moveit_collision_detection.dir/src/allvalid/collision_robot_allvalid.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/mike/ws_acutronics/src/moveit2/moveit_core/collision_detection/include/moveit/collision_detection/collision_common.h:47:0,
                 from /home/mike/ws_acutronics/src/moveit2/moveit_core/collision_detection/include/moveit/collision_detection/collision_matrix.h:40,
                 from /home/mike/ws_acutronics/src/moveit2/moveit_core/collision_detection/include/moveit/collision_detection/collision_robot.h:40,
                 from /home/mike/ws_acutronics/src/moveit2/moveit_core/collision_detection/include/moveit/collision_detection/collision_plugin.h:39,
                 from /home/mike/ws_acutronics/src/moveit2/moveit_core/collision_detection_fcl/include/moveit/collision_detection_fcl/collision_detector_fcl_plugin_loader.h:8,
                 from /home/mike/ws_acutronics/src/moveit2/moveit_core/collision_detection_fcl/src/collision_detector_fcl_plugin_loader.cpp:1:
/home/mike/ws_acutronics/src/moveit2/moveit_core/robot_model/include/moveit/robot_model/robot_model.h:43:10: fatal error: srdfdom/model.h: No such file or directory
 #include <srdfdom/model.h>
          ^~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [collision_detection_fcl/CMakeFiles/collision_detector_fcl_plugin.dir/src/collision_detector_fcl_plugin_loader.cpp.o] Error 1
make[1]: *** [collision_detection_fcl/CMakeFiles/collision_detector_fcl_plugin.dir/all] Error 2
In file included from /home/mike/ws_acutronics/src/moveit2/moveit_core/collision_detection/include/moveit/collision_detection/collision_common.h:47:0,
                 from /home/mike/ws_acutronics/src/moveit2/moveit_core/collision_detection/include/moveit/collision_detection/collision_matrix.h:40,
                 from /home/mike/ws_acutronics/src/moveit2/moveit_core/collision_detection/include/moveit/collision_detection/collision_world.h:42,
                 from /home/mike/ws_acutronics/src/moveit2/moveit_core/collision_detection_fcl/include/moveit/collision_detection_fcl/collision_common.h:41,
[... cut ...]
/usr/include/eigen3/Eigen/src/Geometry/Transform.h:331:5: error: static assertion failed: YOU_PERFORMED_AN_INVALID_TRANSFORMATION_CONVERSION
     EIGEN_STATIC_ASSERT(EIGEN_IMPLIES(OtherMode==int(Affine)||OtherMode==int(AffineCompact), Mode!=int(Isometry)),
     ^
make[2]: *** [transforms/CMakeFiles/moveit_transforms.dir/src/transforms.cpp.o] Error 1
make[1]: *** [transforms/CMakeFiles/moveit_transforms.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< moveit_core	[ Exited with code 2 ]
```

mlautman, 2019-04-16:
I get this error building both locally and in the docker container

vmayoral, 2019-04-16:
@mlautman, thanks for the help. I'm unable to reproduce your error. Instead,  following [the instructions provided above](https://github.com/AcutronicRobotics/moveit2/issues/13#issuecomment-482950340) I get https://gist.github.com/vmayoral/2f2355a3b8cdb242f010a9f6906fe1fd. Just tested now.

>@anasarrak, The build command is in the docker_entrypoint.sh file so anytime you do docker run -it .... it builds the entire project. This could be avoided by having an underlay that is built in the Dockerfile but this is not the most pressing issue

If you wish to debug the environment, simply comment https://github.com/AcutronicRobotics/moveit2/blob/robot-model-tests4/.docker/ci/docker-entrypoint.sh#L11 and launch the build manually. You should get the same result.

dirk-thomas, 2019-04-17:
I can't reproduce the problem with the current steps but see various other problems and unfortunately don't have the bandwidth to iterate on the build process itself.

> it seems that I can see the missing symbols in the local `libmoveit_test_utils library` when inspecting manually.

I would suggest to double check with `VERBOSE=1` if it is actually trying to link against the library where you found the symbol in.

anasarrak, 2019-04-19:
Reproducing the robot model test, I'm having the same errors as @vmayoral 
```bash
Starting >>> moveit_core
--- stderr: moveit_core                             
CMakeFiles/test_robot_model.dir/test/test.cpp.o: In function `LoadPlanningModelsPr2_Model_Test::TestBody()':
test.cpp:(.text+0x4b3): undefined reference to `moveit::core::RobotModel::getJointModel(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
CMakeFiles/test_robot_model.dir/test/test.cpp.o: In function `SiblingAssociateLinks_SimpleYRobot_Test::TestBody()':
test.cpp:(.text+0x83a): undefined reference to `moveit::core::RobotModelBuilder::RobotModelBuilder(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
test.cpp:(.text+0x901): undefined reference to `moveit::core::RobotModelBuilder::addChain(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::vector<geometry_msgs::msg::Pose_<std::allocator<void> >, std::allocator<geometry_msgs::msg::Pose_<std::allocator<void> > > > const&)'
test.cpp:(.text+0x9d7): undefined reference to `moveit::core::RobotModelBuilder::addChain(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::vector<geometry_msgs::msg::Pose_<std::allocator<void> >, std::allocator<geometry_msgs::msg::Pose_<std::allocator<void> > > > const&)'
test.cpp:(.text+0xaad): undefined reference to `moveit::core::RobotModelBuilder::addChain(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::vector<geometry_msgs::msg::Pose_<std::allocator<void> >, std::allocator<geometry_msgs::msg::Pose_<std::allocator<void> > > > const&)'
test.cpp:(.text+0xb83): undefined reference to `moveit::core::RobotModelBuilder::addChain(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::vector<geometry_msgs::msg::Pose_<std::allocator<void> >, std::allocator<geometry_msgs::msg::Pose_<std::allocator<void> > > > const&)'
test.cpp:(.text+0xcac): undefined reference to `moveit::core::RobotModelBuilder::addVirtualJoint(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
test.cpp:(.text+0xdf9): undefined reference to `moveit::core::RobotModelBuilder::addGroup(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
test.cpp:(.text+0xe87): undefined reference to `moveit::core::RobotModelBuilder::isValid()'
test.cpp:(.text+0xfbb): undefined reference to `moveit::core::RobotModelBuilder::build()'
test.cpp:(.text+0x12bf): undefined reference to `moveit::core::RobotModel::getLinkModel(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
CMakeFiles/test_robot_model.dir/test/test.cpp.o: In function `moveit::tools::Profiler::Status(std::ostream&, bool)':
test.cpp:(.text._ZN6moveit5tools8Profiler6StatusERSob[_ZN6moveit5tools8Profiler6StatusERSob]+0x12): undefined reference to `moveit::tools::Profiler::instance()'
test.cpp:(.text._ZN6moveit5tools8Profiler6StatusERSob[_ZN6moveit5tools8Profiler6StatusERSob]+0x28): undefined reference to `moveit::tools::Profiler::status(std::ostream&, bool)'
CMakeFiles/test_robot_model.dir/test/test.cpp.o: In function `LoadPlanningModelsPr2::SetUp()':
test.cpp:(.text._ZN21LoadPlanningModelsPr25SetUpEv[_ZN21LoadPlanningModelsPr25SetUpEv]+0x4e): undefined reference to `moveit::core::loadTestingRobotModel(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [robot_model/test_robot_model] Error 1
make[1]: *** [robot_model/CMakeFiles/test_robot_model.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< moveit_core	[ Exited with code 2 ]

Summary: 0 packages finished [0.96s]
  1 package failed: moveit_core
  1 package had stderr output: moveit_core
```

About the error that you have @mlautman I've just created a clean docker with ros2 crystal and I didn't have any issue with any of the dependencies.

Using these repos https://github.com/AcutronicRobotics/moveit2/blob/master/moveit2.repos you should be able to build them without problem...

anasarrak, 2019-04-19:
Updating the robot_model/CMakelist and leaving it as https://github.com/ros-planning/moveit/blob/7fdc7f35f763c5e6784b5a72665188b2426fc009/moveit_core/robot_model/CMakeLists.txt#L25 the errors has changed to:

```bash
anas@anas:~/github/moveit2$ colcon build --merge-install --packages-select moveit_core
Starting >>> moveit_core
--- stderr: moveit_core                             
../utils/libmoveit_test_utils.a(robot_model_test_utils.cpp.o):(.bss+0x0): multiple definition of `kinematics::LOGGER'
CMakeFiles/test_robot_model.dir/test/test.cpp.o:(.bss+0x0): first defined here
libmoveit_robot_model.a(robot_model.cpp.o):(.bss+0x0): multiple definition of `kinematics::LOGGER'
CMakeFiles/test_robot_model.dir/test/test.cpp.o:(.bss+0x0): first defined here
libmoveit_robot_model.a(robot_model.cpp.o):(.bss+0x10): multiple definition of `moveit::core::LOGGER'
../utils/libmoveit_test_utils.a(robot_model_test_utils.cpp.o):(.bss+0x10): first defined here
libmoveit_robot_model.a(floating_joint_model.cpp.o):(.bss+0x0): multiple definition of `moveit::core::LOGGER'
../utils/libmoveit_test_utils.a(robot_model_test_utils.cpp.o):(.bss+0x10): first defined here
libmoveit_robot_model.a(joint_model_group.cpp.o):(.bss+0x0): multiple definition of `kinematics::LOGGER'
CMakeFiles/test_robot_model.dir/test/test.cpp.o:(.bss+0x0): first defined here
CMakeFiles/test_robot_model.dir/test/test.cpp.o: In function `moveit::tools::Profiler::Status(std::ostream&, bool)':
test.cpp:(.text._ZN6moveit5tools8Profiler6StatusERSob[_ZN6moveit5tools8Profiler6StatusERSob]+0x12): undefined reference to `moveit::tools::Profiler::instance()'
test.cpp:(.text._ZN6moveit5tools8Profiler6StatusERSob[_ZN6moveit5tools8Profiler6StatusERSob]+0x28): undefined reference to `moveit::tools::Profiler::status(std::ostream&, bool)'
../utils/libmoveit_test_utils.a(robot_model_test_utils.cpp.o): In function `moveit::core::loadSRDFModel(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
robot_model_test_utils.cpp:(.text+0xa23): undefined reference to `srdf::Model::initFile(urdf::ModelInterface const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
../utils/libmoveit_test_utils.a(robot_model_test_utils.cpp.o): In function `moveit::core::RobotModelBuilder::RobotModelBuilder(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
robot_model_test_utils.cpp:(.text+0xc0d): undefined reference to `srdf::SRDFWriter::SRDFWriter()'
../utils/libmoveit_test_utils.a(robot_model_test_utils.cpp.o): In function `moveit::core::RobotModelBuilder::build()':
robot_model_test_utils.cpp:(.text+0x3780): undefined reference to `srdf::SRDFWriter::updateSRDFModel(urdf::ModelInterface const&)'
../utils/libmoveit_test_utils.a(robot_model_test_utils.cpp.o): In function `std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count<srdf::SRDFWriter*>(srdf::SRDFWriter*)':
robot_model_test_utils.cpp:(.text._ZNSt14__shared_countILN9__gnu_cxx12_Lock_policyE2EEC2IPN4srdf10SRDFWriterEEET_[_ZNSt14__shared_countILN9__gnu_cxx12_Lock_policyE2EEC5IPN4srdf10SRDFWriterEEET_]+0x56): undefined reference to `srdf::SRDFWriter::~SRDFWriter()'
../utils/libmoveit_test_utils.a(robot_model_test_utils.cpp.o): In function `std::_Sp_counted_ptr<srdf::SRDFWriter*, (__gnu_cxx::_Lock_policy)2>::_M_dispose()':
robot_model_test_utils.cpp:(.text._ZNSt15_Sp_counted_ptrIPN4srdf10SRDFWriterELN9__gnu_cxx12_Lock_policyE2EE10_M_disposeEv[_ZNSt15_Sp_counted_ptrIPN4srdf10SRDFWriterELN9__gnu_cxx12_Lock_policyE2EE10_M_disposeEv]+0x1e): undefined reference to `srdf::SRDFWriter::~SRDFWriter()'
libmoveit_robot_model.a(robot_model.cpp.o): In function `moveit::core::RobotModel::buildModel(urdf::ModelInterface const&, srdf::Model const&)':
robot_model.cpp:(.text+0xad7): undefined reference to `moveit::tools::Profiler::instance()'
robot_model.cpp:(.text+0xaf1): undefined reference to `moveit::tools::Profiler::instance()'
libmoveit_robot_model.a(robot_model.cpp.o): In function `moveit::core::RobotModel::buildJointInfo()':
robot_model.cpp:(.text+0x23b7): undefined reference to `moveit::tools::Profiler::instance()'
robot_model.cpp:(.text+0x23d1): undefined reference to `moveit::tools::Profiler::instance()'
libmoveit_robot_model.a(robot_model.cpp.o): In function `moveit::core::RobotModel::constructShape(urdf::Geometry const*)':
robot_model.cpp:(.text+0x903f): undefined reference to `moveit::tools::Profiler::instance()'
robot_model.cpp:(.text+0x9116): undefined reference to `shapes::Sphere::Sphere(double)'
robot_model.cpp:(.text+0x918d): undefined reference to `shapes::Box::Box(double, double, double)'
robot_model.cpp:(.text+0x91d8): undefined reference to `shapes::Cylinder::Cylinder(double, double)'
robot_model.cpp:(.text+0x925d): undefined reference to `shapes::createMeshFromResource(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, Eigen::Matrix<double, 3, 1, 0, 3, 1> const&)'
libmoveit_robot_model.a(robot_model.cpp.o): In function `moveit::core::RobotModel::getVariableIndex(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const':
robot_model.cpp:(.text+0xa992): undefined reference to `moveit::Exception::Exception(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
[... cut ...]
link_model.cpp:(.text+0x77e): undefined reference to `typeinfo for shapes::Mesh'
link_model.cpp:(.text+0x785): undefined reference to `typeinfo for shapes::Shape'
collect2: error: ld returned 1 exit status
make[2]: *** [robot_model/test_robot_model] Error 1
make[1]: *** [robot_model/CMakeFiles/test_robot_model.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< moveit_core	[ Exited with code 2 ]

Summary: 0 packages finished [4.27s]
  1 package failed: moveit_core
  1 package had stderr output: moveit_core

```

vmayoral, 2019-05-15:
Fixed by https://github.com/AcutronicRobotics/moveit2/pull/71

vmayoral, 2019-05-15:
Thanks everyone that contributed. Specially @dirk-thomas and @mlautman. Credit to @ahcorde and @anasarrak.

closed 2019-05-15 by vmayoral