AcutronicRobotics/moveit2 #19: Isometry3d

vmayoral, 2019-03-03:
@anasarrak, let's go back to Isometry3d please. You have an example at the transforms submodule of moveit_core https://github.com/ros-planning/moveit2/pull/12/commits/94ebbcc191db16d6e3fae760d2d8b8062517d53f.

This follows from https://github.com/ros2/geometry2/pull/93 which changes geometry2. 
**Make sure you pull the `master` branch of geometry2**.

-----

@LanderU, can you please update CI so that we fetch the `master` branch (instead of the `ros2` branch as we were doing before)?

LanderU, 2019-03-03:
Is this enough? https://github.com/AcutronicRobotics/moveit2/commit/de4cf96bd961382be7a65f58e5eb6a7e3c894ecf

vmayoral, 2019-03-03:
That's perfect. Thanks.

vmayoral, 2019-03-03:
@anasarrak, we need to change Isometry3d everywhere. Can you help with it and start from the beginning again ensuring that modules compile while making the changes?

I'd start from `master` branch, comment https://github.com/AcutronicRobotics/moveit2/blob/master/moveit_core/CMakeLists.txt#L178-L204 and go submodule-by-submodule ensuring that things make sense.

vmayoral, 2019-03-03:
@anasarrak fixed all packages except the ones pending of the 3 PRs, please review those three yourself.

vmayoral, 2019-03-03:
For doing so, please go change by change, if you rebase/merge stuff you will end up modifying things in robot_base which you don't want to (unless critically needed).

vmayoral, 2019-03-03:
Things are failing at CI https://travis-ci.org/AcutronicRobotics/moveit2/builds/501241372?utm_source=github_status&utm_medium=notification, maybe the CI isn't properly configured? @anasarrak please validate it tomorrow yourself including the latest changes from other repos described above.

anasarrak, 2019-03-04:
Problems while compiling master:
```bash
/media/erle/myfiles2/github/moveitmaster/src/moveit2/moveit_core/transforms/include/moveit/transforms/transforms.h:51:33: warning: extra ‘;’ [-Wpedantic]
 MOVEIT_CLASS_FORWARD(Transforms);
                                 ^
In file included from /usr/include/eigen3/Eigen/Core:347:0,
                 from /usr/include/eigen3/Eigen/Geometry:11,
                 from /media/erle/myfiles2/github/moveitmaster/src/moveit2/moveit_core/transforms/include/moveit/transforms/transforms.h:42,
                 from /media/erle/myfiles2/github/moveitmaster/src/moveit2/moveit_core/transforms/src/transforms.cpp:37:
/usr/include/eigen3/Eigen/src/Geometry/Transform.h: In instantiation of ‘Eigen::Transform<Scalar, Dim, Mode, _Options>::Transform(const Eigen::Transform<_Scalar, Dim, OtherMode, OtherOptions>&) [with int OtherMode = 2; int OtherOptions = 0; _Scalar = double; int _Dim = 3; int _Mode = 1; int _Options = 0]’:
/media/erle/myfiles2/github/moveitmaster/src/moveit2/moveit_core/transforms/src/transforms.cpp:128:36:   required from here
/usr/include/eigen3/Eigen/src/Geometry/Transform.h:331:5: error: static assertion failed: YOU_PERFORMED_AN_INVALID_TRANSFORMATION_CONVERSION
     EIGEN_STATIC_ASSERT(EIGEN_IMPLIES(OtherMode==int(Affine)||OtherMode==int(AffineCompact), Mode!=int(Isometry)),
```
This error is the same as the one that we were having with the Isometry3d, the way (not the correct way) to solve it, was to change it to Affine3d. 

Exactly the line where it's crashing:
```
Eigen::Isometry3d t = tf2::transformToEigen(transform);
```
I have traied to compare the ros1 and ros2 geometry2 transformToEigen functions, but there is no differences.

Testing if these changes can fix the problem https://github.com/ros/geometry2/pull/335/commits

anasarrak, 2019-03-04:
> Testing if these changes can fix the problem https://github.com/ros/geometry2/pull/335/commits

The problem shouldn't be the geometry2. Focusing more on the transforms since is the one that is causing the problem.

anasarrak, 2019-03-04:
```
add_subdirectory(version)
add_subdirectory(macros)
add_subdirectory(backtrace)
add_subdirectory(exceptions)
add_subdirectory(profiler)
add_subdirectory(logging)
add_subdirectory(background_processing)
add_subdirectory(kinematics_base)
add_subdirectory(controller_manager)
add_subdirectory(sensor_manager)
add_subdirectory(robot_model)
# add_subdirectory(transforms)
# add_subdirectory(robot_state)
# add_subdirectory(robot_trajectory)
add_subdirectory(collision_detection)
add_subdirectory(collision_detection_fcl)
add_subdirectory(kinematic_constraints)
add_subdirectory(planning_scene)
add_subdirectory(constraint_samplers)
add_subdirectory(planning_interface)
add_subdirectory(planning_request_adapter)
add_subdirectory(trajectory_processing)
add_subdirectory(distance_field)
# add_subdirectory(collision_distance_field)
# add_subdirectory(kinematics_metrics)
# add_subdirectory(dynamics_solver)
add_subdirectory(utils)
```

The ones that are failing are **transforms, robot_state and robot_trajectory** (or at least this way it compiles)

anasarrak, 2019-03-04:
Similar case for what they do for the [testing](https://github.com/ros2/geometry2/blob/aaae4cf046e44787be12e4f5209cef34a8123a39/tf2_eigen/test/tf2_eigen-test.cpp#L109) on the t2_eigen
```
  Eigen::Isometry3d I(tm);

  msg = tf2::eigenToTransform(T);
  Eigen::Isometry3d Iback = tf2::transformToEigen(msg);

```

anasarrak, 2019-03-04:
A bit strange behavior, it's supposed that the transformToEigen function to get transformStamped and Transform msgs, but if I send just a transform msg like this:
```
    Eigen::Isometry3d t = tf2::transformToEigen(transform.transform);
```
It does't detect it. 
```
/media/erle/myfiles2/github/moveitmaster/src/moveit2/moveit_core/transforms/src/transforms.cpp: In member function ‘void moveit::core::Transforms::setTransform(const TransformStamped&)’:
/media/erle/myfiles2/github/moveitmaster/src/moveit2/moveit_core/transforms/src/transforms.cpp:131:59: error: invalid initialization of reference of type ‘const TransformStamped& {aka const geometry_msgs::msg::TransformStamped_<std::allocator<void> >&}’ from expression of type ‘const _transform_type {aka const geometry_msgs::msg::Transform_<std::allocator<void> >}’
     Eigen::Isometry3d t = tf2::transformToEigen(transform.transform);
                                                 ~~~~~~~~~~^~~~~~~~~
In file included from /media/erle/myfiles2/github/moveitmaster/src/moveit2/moveit_core/transforms/src/transforms.cpp:38:0:
/home/erle/myfiles/ros2_ws/install/include/tf2_eigen/tf2_eigen.h:46:17: note: in passing argument 1 of ‘Eigen::Affine3d tf2::transformToEigen(const TransformStamped&)’
 Eigen::Affine3d transformToEigen(const geometry_msgs::msg::TransformStamped& t) {
```

On the documentation it says that it can be used on both cases

Eigen::Isometry3d | transformToEigen (const geometry_msgs::Transform &t) 
_Convert a timestamped transform to the equivalent Eigen data type. More..._
 
Eigen::Isometry3d | transformToEigen (const geometry_msgs::TransformStamped &t) _Convert a timestamped transform to the equivalent Eigen data type. More..._

anasarrak, 2019-03-04:
> The ones that are failing are **transforms, robot_state and robot_trajectory** (or at least this way it compiles)

On the 3 cases is breaking for the same transformation, and I noticed that the changes were made by us (since the transform was not getting and we had to change it to transformstamped).

**Robot_state:**
https://github.com/AcutronicRobotics/moveit2/commit/2559cdc50da614295a46921525aed70a94dc10e1#diff-e03d4d985fd8f5e9565368bddcac8d96

**robot_state**:
https://github.com/AcutronicRobotics/moveit2/blob/19923ce1b140eb743a27891aa29207b2d19d00b4/moveit_core/robot_trajectory/src/robot_trajectory.cpp#L402

**transforms**
https://github.com/ros-planning/moveit/blob/211158d00a6938c819f5d51301662b9d993b0c19/moveit_core/transforms/src/transforms.cpp#L127

It should be transform.transform, on our case is just transform

anasarrak, 2019-03-04:
I dont know if the problem is my ws, I'll prepare a docker container to test it while I continue exploring the problem.

vmayoral, 2019-03-04:
No need to prepare a docker container. The CI container works.

anasarrak, 2019-03-04:
Solved the problem, ros2 automatically sources tf2_eigen, so, sourcing again our external dependencies doesn't work, it stays with the preview config. I have changed the ros2_ws/isntall/include/tf2_eigen/tf2_eigen.h to the changes that you've  made to make it work with Isometry3d, and now it seems to compile. The CI is not passing for a similar problem.

LanderU, 2019-03-04:
CI working again: https://travis-ci.org/AcutronicRobotics/moveit2/builds/501604543

vmayoral, 2019-03-04:
Well done guys, closing here then.

closed 2019-03-04 by vmayoral