AcutronicRobotics/moveit2 #27: CI failing with image_common

vmayoral, 2019-03-09:
@LanderU according to https://travis-ci.org/AcutronicRobotics/moveit2/jobs/503697295 we seem to have an issue with `image_common`, right? I guess this comes from a requirement related to https://github.com/AcutronicRobotics/moveit2/pull/26?

LanderU, 2019-03-09:
Yes, I'm working on it.

LanderU, 2019-03-09:
Linux master branch: https://travis-ci.org/AcutronicRobotics/moveit2/jobs/503995410

LanderU, 2019-03-09:
OS X master branch: https://travis-ci.org/AcutronicRobotics/moveit2/jobs/503995411

LanderU, 2019-03-09:
Fixed on master: https://travis-ci.org/AcutronicRobotics/moveit2/builds/503995409, I'm going to fix it on perception branch as well.

LanderU, 2019-03-09:
Fixed on perception for Linux: https://travis-ci.org/AcutronicRobotics/moveit2/jobs/504057073

LanderU, 2019-03-09:
Fixed on perception for OS X: https://travis-ci.org/AcutronicRobotics/moveit2/jobs/504079320

LanderU, 2019-03-09:
Everything working, perception branch status: https://travis-ci.org/AcutronicRobotics/moveit2/builds/504079318

closed 2019-03-09 by LanderU

vmayoral, 2019-03-10:
Thanks a lot @LanderU