AcutronicRobotics/moveit2 #30: OS X: Could NOT find OpenMP_C

vmayoral, 2019-03-15:
@LanderU, I'm going pretty crazy while trying to compile `moveit_ros_perception` in OS X, it goes as follows:
```bash
colcon build --merge-install --packages-select moveit_ros_perception
Starting >>> moveit_ros_perception
--- stderr: moveit_ros_perception
CMake Error at /usr/local/Cellar/cmake/3.13.3/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find OpenMP_C (missing: OpenMP_C_FLAGS OpenMP_C_LIB_NAMES)
Call Stack (most recent call first):
  /usr/local/Cellar/cmake/3.13.3/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  /usr/local/Cellar/cmake/3.13.3/share/cmake/Modules/FindOpenMP.cmake:473 (find_package_handle_standard_args)
  CMakeLists.txt:47 (find_package)

---
Failed   <<< moveit_ros_perception	[ Exited with code 1 ]

Summary: 0 packages finished [9.80s]
  1 package failed: moveit_ros_perception
  1 package had stderr output: moveit_ros_perception
The file /Users/victor/ros2_moveit_ws/None does not exist.
```

Did you encountered this? If so, how did you address it?

LanderU, 2019-03-15:
```bash
brew install libomp
```

vmayoral, 2019-03-15:
Nop, done that already.

LanderU, 2019-03-15:
Which branch are you trying to compile?

LanderU, 2019-03-15:
Checking my commits: https://github.com/AcutronicRobotics/moveit2/pull/26/commits/0f68ad24cdd0a7f9c9427030d1254743decdba6a

vmayoral, 2019-03-15:
Master

LanderU, 2019-03-15:
I just have triggered a build on that branch. If doesn't compile I'll have a look.

https://travis-ci.org/AcutronicRobotics/moveit2/jobs/506958793

vmayoral, 2019-03-15:
Could it be clang-related?

LanderU, 2019-03-15:
Could be, and CI failed because of the actions. I'll review this tomorrow.

vmayoral, 2019-03-16:
Managed to get past this error by:
```bash
victor:~/ros2_moveit_ws$
| => colcon build --merge-install --cmake-args -DOpenMP_CXX_LIB_NAMES="omp" -DOpenMP_CXX_FLAGS="-Xpreprocessor -fopenmp -lomp -I/usr/local/opt/libomp/include" -DOpenMP_omp_LIBRARY="/usr/local/opt/libomp/lib/libomp.dylib" --packages-select moveit_ros_perception
Starting >>> moveit_ros_perception
--- stderr: moveit_ros_perception
In file included from /Users/victor/ros2_moveit_ws/src/moveit2/moveit_ros/perception/occupancy_map_monitor/src/occupancy_map_updater.cpp:37:
/Users/victor/ros2_moveit_ws/src/moveit2/moveit_ros/perception/occupancy_map_monitor/include/moveit/occupancy_map_monitor/occupancy_map_monitor.h:43:10: fatal error: 'pluginlib/class_loader.hpp' file not found
#include <pluginlib/class_loader.hpp>
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /Users/victor/ros2_moveit_ws/src/moveit2/moveit_ros/perception/occupancy_map_monitor/src/occupancy_map_monitor.cpp:41:
/Users/victor/ros2_moveit_ws/src/moveit2/moveit_ros/perception/occupancy_map_monitor/include/moveit/occupancy_map_monitor/occupancy_map_monitor.h:43:10: fatal error: 'pluginlib/class_loader.hpp' file not found
#include <pluginlib/class_loader.hpp>
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
make[2]: *** [occupancy_map_monitor/CMakeFiles/moveit_occupancy_map_monitor.dir/src/occupancy_map_updater.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
1 error generated.
make[2]: *** [occupancy_map_monitor/CMakeFiles/moveit_occupancy_map_monitor.dir/src/occupancy_map_monitor.cpp.o] Error 1
make[1]: *** [occupancy_map_monitor/CMakeFiles/moveit_occupancy_map_monitor.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< moveit_ros_perception	[ Exited with code 2 ]

Summary: 0 packages finished [15.7s]
  1 package failed: moveit_ros_perception
  1 package had stderr output: moveit_ros_perception
The file /Users/victor/ros2_moveit_ws/None does not exist.
```

Source of discussion: https://github.com/Microsoft/LightGBM/issues/1898#issuecomment-455244604. It seems that Mojave is having issues.

closed 2019-03-16 by vmayoral