AcutronicRobotics/moveit2 #33: MoveIt 2 logo

vmayoral, 2019-03-24:
For upcoming announcements, we should use a different logo than the MoveIt! one. @davetcoleman and @mlautman, feel free to add your views and comment on this. 
To accelerate this, @izamalloa, can you please slightly modify 

[image]

with the following requests (_I'll be editing this based on the input we receive_):
1. Replace the `MoveIt!` with `MoveIt 2`

vmayoral, 2019-03-25:
@lalzolak please coordinate this.

izamalloa, 2019-03-25:
Here some prototypes. Once we decide the line to follow, I **will improve it** and create different types of images.
[image]

vmayoral, 2019-03-25:
I'd say we go for the second one @izamalloa but you're the design expert so feel free to make the final choice yourself. Also, can you please ensure that the colours are the exact same ones as the ones in the existing logo (I see the blue color is different). Refer to https://moveit.ros.org/ if you need to review them.

Can you put all the design files in https://drive.google.com/drive/u/2/folders/1n506AcayaQKrdhpnu_-CfORDjWSGsAlq so that we can distribute them if required by the community?

izamalloa, 2019-03-25:
Uploaded the [logo in different formats](https://drive.google.com/open?id=1n506AcayaQKrdhpnu_-CfORDjWSGsAlq).

vmayoral, 2019-03-25:
Great, thanks a lot @izamalloa.

closed 2019-03-25 by vmayoral

robroboto, 2019-03-26:
Hi all, we will be releasing the new official MoveIt logos shortly and publishing them to the MoveIt website. Stay tuned :)

vmayoral, 2019-03-26:
Thanks Rob, let us know when available.

robroboto, 2019-03-27:
Hi all, I've uploaded all the new logos here in the short term: https://drive.google.com/drive/folders/1iOpbikPiZw3ZmjHT42fphujMKzcDWvGk?usp=sharing

We'll add to the MoveIt website in the near future as there are several ongoing website sprints ongoing at the moment.

Thanks!

Rob