AcutronicRobotics/moveit2 #34: CI syncing

vmayoral, 2019-03-24:
@LanderU, there seems to be good progress at https://github.com/ros-planning/moveit_ci and this can certainly complement our simply CI for validating that commits build across our target OSs (OS X and Linux).

Can you please have a look, when possible to https://github.com/ros-planning/moveit_ci#running-locally-for-testing and report here your views on this? It'd be great to get aligned with the MoveIt community on this and re-use the great work they've done on CI over the years.

vmayoral, 2019-03-24:
Note they're using the https://github.com/ros-planning/moveit2/tree/moveit2-ci branch instead.

LanderU, 2019-03-24:
Alright, I'll have a look.

LanderU, 2019-03-24:
```bash
lander@wks4:/tmp/travis/moveit2(master)$ .moveit_ci/travis.sh
Testing branch 'master' of 'moveit2' on ROS ''
Starting Docker image: moveit/moveit2:-ci
docker pull moveit/moveit2:-ci
invalid reference format
travis_time:end:0194a45c:start=1553455316983103808,finish=1553455317017281051,duThe command "docker pull moveit/moveit2:-ci" failed with error code 1.
```

LanderU, 2019-03-24:
It seems that some of the packages are not ready for rosdep yet:

```bash
travis_time:end:2f2e3248:start=1553457928622975850,finish=1553457928628634733,durosdep install -y -q -n --from-paths . --ignore-src --rosdistro crystal
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
moveit_ros_benchmarks: Cannot locate rosdep definition for [roscpp]
moveit_planners_chomp: Cannot locate rosdep definition for [rostest]
moveit_planners_ompl: Cannot locate rosdep definition for [rosunit]
moveit_controller_manager_example: Cannot locate rosdep definition for [roscpp]
moveit_ros_planning: Cannot locate rosdep definition for [srdfdom]
moveit_experimental: Cannot locate rosdep definition for [catkin]
moveit_plugins: Cannot locate rosdep definition for [catkin]
moveit_ros_visualization: Cannot locate rosdep definition for [rostest]
moveit_ros_planning_interface: Cannot locate rosdep definition for [rostest]
moveit_setup_assistant: Cannot locate rosdep definition for [rosunit]
moveit_runtime: Cannot locate rosdep definition for [catkin]
moveit_fake_controller_manager: Cannot locate rosdep definition for [roscpp]
moveit_kinematics: Cannot locate rosdep definition for [xmlrpcpp]
moveit_ros: Cannot locate rosdep definition for [catkin]
moveit_ros_robot_interaction: Cannot locate rosdep definition for [rosunit]
moveit_ros_move_group: Cannot locate rosdep definition for [moveit_resources]
chomp_motion_planner: Cannot locate rosdep definition for [catkin]
moveit_ros_control_interface: Cannot locate rosdep definition for [controller_manager_msgs]
moveit_simple_controller_manager: Cannot locate rosdep definition for [actionlib]
moveit_chomp_optimizer_adapter: Cannot locate rosdep definition for [catkin]
moveit_planners: Cannot locate rosdep definition for [catkin]
moveit_core: Cannot locate rosdep definition for [code_coverage]
moveit_ros_perception: Cannot locate rosdep definition for [rosunit]
moveit_ros_warehouse: Cannot locate rosdep definition for [rosconsole]
moveit_commander: Cannot locate rosdep definition for [moveit_resources]
moveit_ros_manipulation: Cannot locate rosdep definition for [rosconsole]
moveit: Cannot locate rosdep definition for [catkin]
travis_time:end:39a4e0c8:start=1553457928632391336,finish=1553457930114174897,duThe command "rosdep install -y -q -n --from-paths . --ignore-src --rosdistro crystal" failed with error code 1. 

Travis script finished with errors.
```

vmayoral, 2019-04-06:
Looking more closely into https://github.com/ros-planning/moveit_ci/tree/ros2:
- We'll need to figure out how to add OS X to their existing setup https://github.com/ros-planning/moveit2/blob/master/.travis.yml
  - [Their .travis.yaml](https://github.com/ros-planning/moveit2/blob/master/.travis.yml)
  - [Ours](https://github.com/AcutronicRobotics/moveit2/blob/master/.travis.yml)
- What our CI does
  - Travis picks both OS X and Linux (Ubuntu Bionic)
  - Depending on the OS, we launch one script or the other
     - for Linux:
         - builds docker image (Bionic) which configures Ubuntu with ROS 2
         - mounts moveit2 directory in ROS 2 workspace and launches a script which builds the complete moveit ws
     - for OS X:
         - fetches the dependencies https://github.com/AcutronicRobotics/moveit2/blob/master/osx-dependencies.sh
         - launches https://github.com/AcutronicRobotics/moveit2/blob/master/osx-compile.sh which installs ROS 2 from the binaries provided by Open Robotics and then puts together a ws for moveit after which, it compiles it.
- what [their](https://github.com/ros-planning/moveit2/blob/master/.travis.yml)  CI does:
  - for Linux:
     - configure a number of env variables
     - clone moveit_ci as `.moveit_ci` inside of the moveit repository
     - launch `.moveit_ci/travis.sh` which (roughly):
        - launches Docker through run_docker
        - update the system to have the right dependencies
        - create a ws and use .rosinstall files to fetch repos
        - build ws
        - run tests

Listing below a list of desires to align ourselves with this ci system:
- [x] Inspect the official CI of moveit
- [ ] Get a fork a moveit_ci and enable it in our ci branch
    - [ ] When valid, bring it to master
- [ ] Run moveit_ci in OS X (with Docker, as in Linux)
- [ ] Run moveit_ci in OS X (natively)
- [ ] Enable OS X in Travis

vmayoral, 2019-04-07:
Stumbled repeatedly on while enabling any moveit "real" package:

```bash
root@c12841dd216b:/opt/ws_moveit# colcon build
Starting >>> moveit_core
--- stderr: moveit_core
CMake Error at /usr/share/cmake-3.10/Modules/FindPkgConfig.cmake:415 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindPkgConfig.cmake:593 (_pkg_check_modules_internal)
  CMakeLists.txt:26 (pkg_check_modules)
```
(e.g. https://travis-ci.org/AcutronicRobotics/moveit2/builds/516620511) 

Issues seems to be in the docker image of moveit https://hub.docker.com/r/moveit/moveit2/dockerfile

vmayoral, 2019-04-07:
@LanderU, can you please build at https://hub.docker.com/u/acutronicrobotics our docker images to accelerate the process (https://github.com/AcutronicRobotics/moveit2)?

LanderU, 2019-04-07:
> @LanderU, can you please build at https://hub.docker.com/u/acutronicrobotics our docker images to accelerate the process (https://github.com/AcutronicRobotics/moveit2)?
Edit: it'd be great if we could set also hooks so that Dockerhub builds them automatically as well.

WIP

LanderU, 2019-04-07:
> Travis picks both OS X and Linux (Ubuntu Bionic)

This is no true at all, I set Ubuntu Bionic but the truth is travis only have support for Xenial.
https://docs.travis-ci.com/user/reference/overview/#virtualization-environments

> Run moveit_ci in OS X (with Docker, as in Linux)

AFAIK this is not possible, there is no docker OS X images, this is why we're running a native machine in travis to build the OS X work space.

vmayoral, 2019-04-07:
> This is no true at all, I set Ubuntu Bionic but the truth is travis only have support for Xenial.
https://docs.travis-ci.com/user/reference/overview/#virtualization-environments

That's right, my bad here.

> AFAIK this is not possible, there is no docker OS X images, this is why we're running a native machine in travis to build the OS X work space.

No, here you misunderstood me. I mean, using the Docker container (the Linux container) from OS X to build unit tests. This is a WIP i'm almost there.

> > @LanderU, can you please build at https://hub.docker.com/u/acutronicrobotics our docker images to accelerate the process (https://github.com/AcutronicRobotics/moveit2)?
> Edit: it'd be great if we could set also hooks so that Dockerhub builds them automatically as well.
> 
> WIP
> 
> 

This is great, it's what I need. It'd be also great to get the credentials of this account, please send them to me privately so that we can co-manage it.

LanderU, 2019-04-07:
> No, here you misunderstood me. I mean, using the Docker container (the Linux container) from OS X to build unit tests. This is a WIP i'm almost there.

That's fantastic, sorry.

> This is great, it's what I need. It'd be also great to get the credentials of this account, please send them to me privately so that we can co-manage it.

Sure!

vmayoral, 2019-04-07:
Merged https://github.com/AcutronicRobotics/moveit2/pull/43

vmayoral, 2019-04-07:
Still failing the tests for some weird matter (though I fixed the issue locally with urdfdom_py, looking into it again)

vmayoral, 2019-04-08:
Process finished and CI's have been aligned. An attempt to integrate our efforts upstream was submitted to https://github.com/ros-planning/moveit_ci/pull/56.

closed 2019-04-08 by vmayoral