AcutronicRobotics/moveit2 #48: CI broken

vmayoral, 2019-04-13:
It appears the CI is broken in master for OS X:
```bash
/usr/local/include/console_bridge/console.h:67:88: warning: token pasting of ',' and __VA_ARGS__ is a GNU extension [-Wgnu-zero-variadic-macro-arguments]
/usr/local/include/console_bridge/console.h:67:88: warning: token pasting of ',' and __VA_ARGS__ is a GNU extension [-Wgnu-zero-variadic-macro-arguments]
/usr/local/include/console_bridge/console.h:67:88: warning: token pasting of ',' and __VA_ARGS__ is a GNU extension [-Wgnu-zero-variadic-macro-arguments]
/usr/local/include/console_bridge/console.h:67:90: warning: token pasting of ',' and __VA_ARGS__ is a GNU extension [-Wgnu-zero-variadic-macro-arguments]
  console_bridge::log(__FILE__, __LINE__, console_bridge::CONSOLE_BRIDGE_LOG_ERROR, fmt, ##__VA_ARGS__)
                                                                                         ^
/usr/local/include/console_bridge/console.h:67:90: warning: token pasting of ',' and __VA_ARGS__ is a GNU extension [-Wgnu-zero-variadic-macro-arguments]
50 warnings generated.
---
Finished <<< srdfdom [2min 19s]
[Processing: geometric_shapes, image_transport, object_recognition_msgs, octomap_msgs]
--- stderr: image_transport
/Users/victor/ros2_moveit_ws/src/image_common/image_transport/test/test_remapping.cpp:22:13: error: no type named 'NodeOptions' in namespace 'rclcpp'
    rclcpp::NodeOptions node_options;
    ~~~~~~~~^
1 error generated.
make[2]: *** [CMakeFiles/image_transport-remapping.dir/test/test_remapping.cpp.o] Error 1
make[1]: *** [CMakeFiles/image_transport-remapping.dir/all] Error 2
make: *** [all] Error 2
make: INTERNAL: Exiting with 9 jobserver tokens available; should be 8!
---
Failed   <<< image_transport	[ Exited with code 2 ]
--- stderr: geometric_shapes
geometric_shapes: You did not request a specific build type: Choosing 'Release' for maximum performance
---
Aborted  <<< geometric_shapes
Aborted  <<< octomap_msgs
Aborted  <<< object_recognition_msgs
```

And also in Linux https://travis-ci.org/AcutronicRobotics/moveit2/builds/519581189#L2586

vmayoral, 2019-04-13:
https://github.com/ros2/tinydir_vendor and several other core ros2 packages had to be updated

LanderU, 2019-04-13:
This is a known issue. I'm going to fix it.

vmayoral, 2019-04-13:
This is just an intuition but I believe the error comes from geometry2 package. We should use https://github.com/AcutronicRobotics/geometry2/commits/ros2 instead.

LanderU, 2019-04-13:
> This is just an intuition but I believe the error comes from geometry2 package. We should use https://github.com/AcutronicRobotics/geometry2/commits/ros2 instead.

Is not working for me. I'm still trying to fix it locally.

LanderU, 2019-04-13:
The errors are not the same.

LanderU, 2019-04-13:
Fixed locally.

vmayoral, 2019-04-13:
I’m still fighting with OS X, getting closer I think

LanderU, 2019-04-13:
> /Users/victor/ros2_moveit_ws/src/image_common/image_transport/test/test_remapping.cpp:22:13: error: no type named 'NodeOptions' in namespace 'rclcpp'
    rclcpp::NodeOptions node_options;

This issue is because you're using a wrong version of image_transport. Make sure that you're using crystal release.

LanderU, 2019-04-13:
We really need to fix the tests part of each package. The error that I'm having is related to linking a test.

LanderU, 2019-04-14:
Master branch working again. https://travis-ci.org/AcutronicRobotics/moveit2/builds/519853092

closed 2019-04-14 by LanderU

vmayoral, 2019-04-14:
@LanderU I just run the CI locally in OS X and obtaining the following:

```bash
Summary: 26 packages finished [1min 2s]
  2 packages had stderr output: octomap tf2_ros
  1 package had test failures: tf2_ros
colcon test-results
Test log of build/tf2_ros/test_results/tf2_ros/tf2_ros_test_message_filter.gtest.xml
<?xml version="1.0" encoding="UTF-8"?>
<testsuite name="tf2_ros" tests="1" failures="1" time="0" errors="0" skip="0">
  <testcase classname="tf2_ros" name="tf2_ros_test_message_filter.gtest.missing_result" status="run" time="0">
    <failure message="The test did not generate a result file."/>
  </testcase>
</testsuite>
build/tf2_ros/test_results/tf2_ros/tf2_ros_test_message_filter.gtest.xml: 1 test, 0 errors, 1 failure, 0 skipped

Summary: 54 tests, 0 errors, 1 failure, 0 skipped

Travis script finished with errors.
```

LanderU, 2019-04-14:
In our master branch we've this package on the blacklist: https://github.com/AcutronicRobotics/moveit2/blob/master/.travis.yml#L22

vmayoral, 2019-04-14:
got it, thanks