AcutronicRobotics/moveit2 #50: test_robot_model failed

LanderU, 2019-04-13:
```bash
CMakeFiles/test_robot_model.dir/test/test.cpp.o: In function `LoadPlanningModelsPr2_Model_Test::TestBody()':
test.cpp:(.text+0x4b3): undefined reference to `moveit::core::RobotModel::getJointModel(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
CMakeFiles/test_robot_model.dir/test/test.cpp.o: In function `SiblingAssociateLinks_SimpleYRobot_Test::TestBody()':
test.cpp:(.text+0x83a): undefined reference to `moveit::core::RobotModelBuilder::RobotModelBuilder(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
test.cpp:(.text+0x901): undefined reference to `moveit::core::RobotModelBuilder::addChain(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::vector<geometry_msgs::msg::Pose_<std::allocator<void> >, std::allocator<geometry_msgs::msg::Pose_<std::allocator<void> > > > const&)'
test.cpp:(.text+0x9d7): undefined reference to `moveit::core::RobotModelBuilder::addChain(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::vector<geometry_msgs::msg::Pose_<std::allocator<void> >, std::allocator<geometry_msgs::msg::Pose_<std::allocator<void> > > > const&)'
test.cpp:(.text+0xaad): undefined reference to `moveit::core::RobotModelBuilder::addChain(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::vector<geometry_msgs::msg::Pose_<std::allocator<void> >, std::allocator<geometry_msgs::msg::Pose_<std::allocator<void> > > > const&)'
test.cpp:(.text+0xb83): undefined reference to `moveit::core::RobotModelBuilder::addChain(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::vector<geometry_msgs::msg::Pose_<std::allocator<void> >, std::allocator<geometry_msgs::msg::Pose_<std::allocator<void> > > > const&)'
test.cpp:(.text+0xcac): undefined reference to `moveit::core::RobotModelBuilder::addVirtualJoint(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
test.cpp:(.text+0xdf9): undefined reference to `moveit::core::RobotModelBuilder::addGroup(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
test.cpp:(.text+0xe87): undefined reference to `moveit::core::RobotModelBuilder::isValid()'
test.cpp:(.text+0xfbb): undefined reference to `moveit::core::RobotModelBuilder::build()'
test.cpp:(.text+0x12bf): undefined reference to `moveit::core::RobotModel::getLinkModel(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
CMakeFiles/test_robot_model.dir/test/test.cpp.o: In function `moveit::tools::Profiler::Status(std::ostream&, bool)':
test.cpp:(.text._ZN6moveit5tools8Profiler6StatusERSob[_ZN6moveit5tools8Profiler6StatusERSob]+0x12): undefined reference to `moveit::tools::Profiler::instance()'
test.cpp:(.text._ZN6moveit5tools8Profiler6StatusERSob[_ZN6moveit5tools8Profiler6StatusERSob]+0x28): undefined reference to `moveit::tools::Profiler::status(std::ostream&, bool)'
CMakeFiles/test_robot_model.dir/test/test.cpp.o: In function `LoadPlanningModelsPr2::SetUp()':
test.cpp:(.text._ZN21LoadPlanningModelsPr25SetUpEv[_ZN21LoadPlanningModelsPr25SetUpEv]+0x4e): undefined reference to `moveit::core::loadTestingRobotModel(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
collect2: error: ld returned 1 exit status
```

LanderU, 2019-04-13:
This issue is related to #48 I can fix it avoiding to compile the tests, but I prefer don't do this. @anasarrak please have a look.

vmayoral, 2019-04-13:
Connected to #13 @LanderU, I've been reviewing this for a while. No solution so far.

https://github.com/AcutronicRobotics/moveit2/commit/64ea5afe0ac108795a69b1ad29ccf64c9e101d19 enables what's requested but still failing.

LanderU, 2019-04-13:
https://travis-ci.org/AcutronicRobotics/moveit2/builds/519751376. I've updated the docker images for CI and looks like it fix the issue.

LanderU, 2019-04-13:
Tomorrow I'm going to do more tests and see if is fully fixed.

LanderU, 2019-04-14:
It looks like the problem is fixed.

closed 2019-04-14 by LanderU