AcutronicRobotics/moveit2 #78: Actions headers doesn't generate properly

LanderU, 2019-05-18:
I'm getting the following errors:

```bash
--- stderr: moveit_simple_controller_manager
In file included from /root/moveit2/moveit_plugins/moveit_simple_controller_manager/src/follow_joint_trajectory_controller_handle.cpp:38:0:
/root/moveit2/moveit_plugins/moveit_simple_controller_manager/include/moveit_simple_controller_manager/follow_joint_trajectory_controller_handle.h:42:10: fatal error: control_msgs/action/follow_joint_trajectory.hpp: No such file or directory
 #include <control_msgs/action/follow_joint_trajectory.hpp>
````

We faced similar issue with HRIM. I'm going to have a look there for a solution.

ahcorde, 2019-05-18:
We need to compile this package https://github.com/ros-controls/control_msgs/tree/crystal-devel

LanderU, 2019-05-19:
I cannot advance as desired, some of the code included in master has dashing stuff and the CI is trying to compile against crystal. I've fixed the problem with the headers using the repo described above.

closed 2019-05-19 by LanderU

vmayoral, 2019-05-19:
Fixed by https://github.com/AcutronicRobotics/moveit2/commit/ab753f97d05670c362e638529e53b0cc786eccab