AcutronicRobotics/moveit2 #79: Unable to compile tf2_ros dashing

LanderU, 2019-05-19:
```bash
In file included from /opt/ws_moveit/src/geometry2/tf2_ros/src/static_transform_broadcaster.cpp:36:0:
/opt/ws_moveit/src/geometry2/tf2_ros/include/tf2_ros/static_transform_broadcaster.h: In instantiation of ‘tf2_ros::StaticTransformBroadcaster::StaticTransformBroadcaster(rclcpp::node_interfaces::NodeTopicsInterface::SharedPtr) [with AllocatorT = std::allocator<void>; rclcpp::node_interfaces::NodeTopicsInterface::SharedPtr = std::shared_ptr<rclcpp::node_interfaces::NodeTopicsInterface>]’:
/opt/ws_moveit/src/geometry2/tf2_ros/src/static_transform_broadcaster.cpp:41:63:   required from here
/opt/ws_moveit/src/geometry2/tf2_ros/include/tf2_ros/static_transform_broadcaster.h:67:76: error: no matching function for call to ‘create_publisher<MessageT, std::allocator<void>, PublisherT>(std::__shared_ptr<rclcpp::node_interfaces::NodeTopicsInterface, (__gnu_cxx::_Lock_policy)2>::element_type*, const char [11], rmw_qos_profile_t&, bool, std::shared_ptr<std::allocator<void> >)’
     publisher_ = rclcpp::create_publisher<MessageT, AllocatorT, PublisherT>(
                  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
       node_topics_interface.get(), "/tf_static",
       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~                            
       custom_qos_profile, false, std::make_shared<AllocatorT>());
```

LanderU, 2019-05-20:
Solved using the pre-installed version.

closed 2019-05-20 by LanderU

ahcorde, 2019-05-20:
This preinstalled version should not be enough

LanderU, 2019-05-20:
I'm reworking the current Docker image to be able to compile the current master against dashing. We should have a look in our port, the status of this certain package.