AcutronicRobotics/moveit2 #81: How to build AcutronicRobotics/moveit2

davetcoleman, 2019-05-20:
How does one build this repo? No one at PickNik has ever been able to compile any version of moveit2. I'm trying to dig into why, and its not clear to me where to start. 

Looking at the README of this repo, almost every section says deprecated:

> Install instructions (DEPRECATED, see below)
> Using a Docker container (DEPRECATED)
> From sources (DEPRECATED)

The only section that may work is to run locally moveit_ci and the Travis script? This is not the purpose of Travis...

vmayoral, 2019-05-21:
> How does one build this repo? No one at PickNik has ever been able to compile any version of moveit2.

Well, I'm sorry for that @davetcoleman. Frankly, I haven't seen a single issue _describing_ specific issues from you or your team members. We'd be happy to help you get up to speed if you share your particular issues about compiling moveit2. Feel free to share them here please.

I take there's room for improvement, indeed and one could say that at this point some basic knowhow of ROS 2 is required to get things up to speed. If you want to quickly get something going without much prior knowledge, our team prepared last week https://github.com/AcutronicRobotics/moveit2/tree/local-build/.docker/local-build specifically for PickNik. @LanderU, can you confirm this is still building nicely?

Alternatively, you could try doing things from source. A good way to start is by looking at how CI does things (currently failing but we're on it and should be addressed shortly). E.g. look at https://github.com/AcutronicRobotics/moveit2/blob/master/.docker/ci/Dockerfile#L13-L64 if you're working in Ubuntu Bionic.

> The only section that may work is to run locally moveit_ci and the Travis script? This is not the purpose of Travis...

I think you're right. We aren't very happy with the current paths but code is changing rapidly and referring to Docker containers (not so much Travis tbh) is the easiest to maintain so far.

To further simplify things, our team is cooking a Docker image with the current version of the code compiled so that everyone can test some simple examples without hurdles.

ahcorde, 2019-05-21:
Hello @davetcoleman ,

I have created a new branch called `master_compile` that I think you will be able to compile it. I ignored some of the packages that create some issues and updated some dependencies.

# Instructions

## Install  ros2 dashing pre-release 

Follow [this](https://discourse.ros.org/t/ros-2-dashing-diademata-call-for-testing-and-package-releases/8819) to install ros2 dashing pre-release 

**NOTE**: Remove tf2 if you installed it from sources
```
sudo apt-get purge ros-dashing-tf2*
```
## Compile moveit2 and dependencies:

```bash
mkdir -p ~/moveit2_ws/src
cd ~/moveit2_ws/src
git clone https://github.com/AcutronicRobotics/moveit2 -b master_compile
cd ~/moveit2_ws
vcs import src < src/moveit2/moveit2.repos
colcon build --merge-install --cmake-args -DBUILD_TESTING=FALSE
```

you will be able to compile the following packages:
 - `moveit_core`
 - `moveit_ros_perception`
 - `moveit_ros_planning`
 - `moveit_fake_controller_manager`
 - `moveit_planners_ompl`

Let me know if you have any issues trying to compile this.

davetcoleman, 2019-05-21:
@ahcorde thanks for those build from source notes, they worked for me! Can you update the README to reflect this latest approach? What's keeping the master branch from building like this special branch does? 

Note after installing ROS 2, I still had to install these packages before I could build the rest of the repo:

> sudo apt-get install python-vcstool python3-colcon-*

I think the install instructions should reflect this.

> Frankly, I haven't seen a single issue describing specific issues from you or your team members.

This is a specific issue requesting assistance in where to find documentation for compiling Acutronic's moveit2 from source.  I very clearly quoted your README and how it could confuse outside developers like myself. 

@vmayoral Thank you for sharing the ``local-build`` branch, perhaps this could be added to the README as the new recommended approach to building from Docker? Currently the documentation is outdated, which makes it hard for collaborators like PickNik to provide support.

> A good way to start is by looking at how CI does things (currently failing but we're on it and should be addressed shortly)

The CI does not seem to work - most of the PRs in this repo are merged without passing CI.  This is very bad software engineering practice.

vmayoral, 2019-05-21:
> @vmayoral Thank you for sharing the local-build branch, perhaps this could be added to the README as the new recommended approach to building from Docker? Currently the documentation is outdated, which makes it hard for collaborators like PickNik to provide support.

Mmmm that sounds odd, I took them from the README. Particularly from https://github.com/AcutronicRobotics/moveit2#using-a-docker-container-deprecated.

> The CI does not seem to work - most of the PRs in this repo are merged without passing CI. This is very bad software engineering practice.

I think this is a valid criticism. The intention is indeed to respect good practices. The transition to Dashing and the global changes required for fixing unit tests have made us twist things quite a bit i think.

For what I can see, CI broke 6 days ago when we started moving things to Dashing and introducing the first demonstrator around f2a18ce02d9d22dd2358ec22175fb5d8a56d49c4. We're working on this as we speak at https://github.com/AcutronicRobotics/moveit2/issues/76. 

> @ahcorde thanks for those build from source notes, they worked for me! Can you update the README to reflect this latest approach? What's keeping the master branch from building like this special branch does?

bab54b9e0d8f1951d596bbd50bea8eda7a55f35f and 8a9e7e28190b21aa576c1571b0923242154a924b update it. Glad it worked for you @davetcoleman.

Closing here then.

closed 2019-05-21 by vmayoral

davetcoleman, 2019-05-21:
> I took them from the README. Particularly from https://github.com/AcutronicRobotics/moveit2#using-a-docker-container-deprecated.

I think our definition of "deprecated" is different. When I see deprecated, I read it to mean "do not use" and "probably does not work". We need build instructions that are not deprecated if we want the community to trust this code.

> For what I can see, CI broke 6 days ago 

[This is not true](https://github.com/AcutronicRobotics/moveit2/pulls?utf8=%E2%9C%93&q=is%3Apr+is%3Aclosed+is%3Amerged), many of the merged PRs going back for months have failed.

Thanks for quickly updating the README with my suggestions and @ahcorde's updates

vmayoral, 2019-05-22:
> This is not true, many of the merged PRs going back for months have failed.

That's right, many failed and we still accepted them to move forward faster. Specifically, for the latest PRs. We should improve this but I still hold my last point with regard Dashing.

I will also point out that there has been historical efforts to comply with good practices. See https://github.com/AcutronicRobotics/moveit2/pull/60, https://github.com/AcutronicRobotics/moveit2/pull/58 or https://github.com/AcutronicRobotics/moveit2/pull/57 among many others. We changed CI twice (due to community requests, actually this was an active topic of discussion @davetcoleman) and simply reviewing the commit history shows how much effort @LanderU among others in our team have put into fixing CI, periodically. 

I'm not justifying the not-properly-reviewed PRs. We should also do this in our fork. I'm simply pointing out that we've constantly been trying to reach a tradeoff between moving forward with the port and peer-reviewed contributions (which still, pass through a layer of review when submitting them upstream). This tradeoff is the reason why we relaxed things in several PRs.

Nevertheless, point noted. Let's try to improve things together. Thanks for the support @davetcoleman.