AcutronicRobotics/moveit2 #93: Error while installing moveit2 from source

mlautman, 2019-05-29:
When I tried to build the `master_compile` branch from source using the README instructions, I am seeing a number of issues. 

1. The distinction between the `master` branch and `master_compile` branch is unclear. Before announcing the progress, it is very important that users are able to reliably work of a `master` branch

2. running `rosdep install -y --from-paths src --ignore-src --rosdistro ${ROS_DISTRO}` to ensure I have the correct dependencies fails with:
```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
moveit_planners_ompl: Cannot locate rosdep definition for [ompl]
```

3. This causes the build to fail with
```
--- stderr: moveit_planners_ompl                                                            
CMake Warning at /home/mike/ws_acutronics/install/share/moveit_core/cmake/ament_cmake_export_libraries-extras.cmake:116 (message):
  Package 'moveit_core' exports library 'moveit_test_utils' which couldn't be
  found
Call Stack (most recent call first):
  /home/mike/ws_acutronics/install/share/moveit_core/cmake/moveit_coreConfig.cmake:38 (include)
  CMakeLists.txt:13 (find_package)

CMake Error at CMakeLists.txt:18 (find_package):
  By not providing "Findompl.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "ompl", but
  CMake did not find one.

  Could not find a package configuration file provided by "ompl" with any of
  the following names:

    omplConfig.cmake
    ompl-config.cmake

  Add the installation prefix of "ompl" to CMAKE_PREFIX_PATH or set
  "ompl_DIR" to a directory containing one of the above files.  If "ompl"
  provides a separate development package or SDK, be sure it has been
  installed.
```

LanderU, 2019-05-29:
> The distinction between the master branch and master_compile branch is unclear. Before announcing the progress, it is very important that users are able to reliably work of a master branch

+1, we'll update the README.md to clarify this point.

> running rosdep install -y --from-paths src --ignore-src --rosdistro ${ROS_DISTRO} to ensure I have the correct dependencies fails with:

You're right this package is failing when you try to install using `rosdep`. You need to install `libompl-dev` using apt. We need to reflect this on the README.md as well.

We'll review the current status of the README.md to clarify all of this points. Thanks for your feedback!

vmayoral, 2019-05-29:
Thanks for reporting @mlautman, @LanderU once fixed, can we refer in here to the commit that addresses the issues raised here please?

@mlautman, based on our experience, rosdep is discouraged at this point (@LanderU, correct me if I'm wrong). AFAIK, we've done a reasonable amount of work to try to fix deps that way but more effort is required. @mlautman feel free to jump in and help with rosdep if you consider this critical, I believe you guys have plenty of experience.

mlautman, 2019-05-29:
> You're right this package is failing when you try to install using rosdep. You need to install libompl-dev using apt. We need to reflect this on the README.md as well.

>  based on our experience, rosdep is discouraged at this point

I don't understand this viewpoint. ROS2 uses rosdep extensively. Why wouldn't we use it here as well? Explicitly installing dependencies with apt is a short-term patch that doesn't actually achieve our goals for MoveIt 2. When we go to release a Debian package we will need to use rosdep.

> AFAIK, we've done a reasonable amount of work to try to fix deps that way but more effort is required

What has been tried? In order to use rosdep with ompl, we need to add a Dashing track for ompl with `bloom-release`. This will add ompl to [distribution.yaml](https://github.com/ros/rosdistro/blob/master/dashing/distribution.yaml) which acts as a dictionary for rosdep.  I have ping'd Mark Moll to see if he can un-block us there.

davetcoleman, 2019-05-30:
I had this same issue. I've made some suggestions:

https://github.com/AcutronicRobotics/moveit2/pull/95

mlautman, 2019-05-30:
On second thought, if you are able to successfully build against `libompl-dev` we don't need to release ompl with bloom. I opened a [PR to ros/distro](https://github.com/ros/rosdistro/compare/master...mlautman:patch-2) that will enable us to install `libompl-dev` using the `libompl-dev` key in package.xml

vmayoral, 2019-06-11:
@LanderU can we close this issue?

closed 2019-06-12 by LanderU