AcutronicRobotics/moveit2 #100: Errors when running ubuntu-install.sh

mlanting, 2019-06-03:
### Description

Following the instructions given here: https://acutronicrobotics.com/docs/products/robots/mara/moveit2/install/ubuntu, the build phase fails with a number of errors having to do with types or functions in rclcpp not being found (see attachment for full build output), such as rclcpp::QoS.

### Your environment
ROS Distro: Crystal (installed by the script itself)
OS Version: Ubuntu 18.04
Branch: master

### Steps to reproduce
1. Clone the repository
2. run `bash ubuntu-install.sh` from within the repository.
3. ???

### Expected behaviour
The script should install moveit2 dependencies and successfully build the workspace

### Actual behaviour
Dependencies are installed, and workspace projects are cloned, but the build fails with errors.

### Backtrace or Console output
[error_output.txt](https://github.com/AcutronicRobotics/moveit2/files/3248200/error_output.txt)

LanderU, 2019-06-03:
Hello @mlanting, sorry, the script uses crystal release of ROS 2, and the code was changed to be compliant with dashing. We'll fix this issue and update the ubuntu-install.sh script to be compliant with the new release. On the other hand, you can use the instructions available on the README.md in the section install and test options --> Build From Source.

Thanks for reporting!

vmayoral, 2019-06-03:
That’s right. Install from that script is deprecated.

Refer to the build from source section.

mlanting, 2019-06-03:
Thanks for the response. I'll take a look at the instructions there then.

closed 2019-06-03 by mlanting