AcutronicRobotics/moveit2 #104: Build error: Package 'moveit_core' exports the library 'moveit_exceptions' which couldn't be found

mlanting, 2019-06-05:
### Description

I'm getting build errors when building from source according to the instructions found here: https://github.com/acutronicrobotics/moveit2#build-from-source

### Your environment
* ROS Distro: Dashing (master branch)
* OS Version: Ubuntu 18.04
* Source build from master branch

### Steps to reproduce
1. Create a new ros2 workspace (~/ros2_ws) and install Dashing from source according to instructions found here: https://index.ros.org/doc/ros2/Installation/Dashing/Linux-Development-Setup/ and using the version of [ros2.repos](https://raw.githubusercontent.com/ros2/ros2/release-latest/ros2.repos) from the master branch. I used the command `colcon build --merge-install --symlink-install --cmake-force-configure` to build ros2.
2. Create a new workspace folder and src folder (~/ros2_moveit_ws/src) and clone the moveit2 repository with the command `git clone https://github.com/AcutronicRobotics/moveit2 -b master` into the src directory.
3. Move to ~/ros2_moveit_ws/ and run the following commands:
~~~~
. ~/ros2_ws/install/setup.bash
vcs import src < src/moveit2/moveit2.repos
colcon build --merge-install --symlink-install --cmake-args -DBUILD_TESTING=FALSE
~~~~
### Expected behaviour
The colcon build command should complete successfully

### Actual behaviour
Build fails on the `moveit_ros_perception` component because it can't find a library exported by moveit_core:
~~~~
Starting >>> moveit_ros_perception
--- stderr: moveit_ros_perception                            
CMake Error at /home/mlanting/ros2_moveit_ws/install/share/moveit_core/cmake/ament_cmake_export_libraries-extras.cmake:48 (message):
  Package 'moveit_core' exports the library 'moveit_exceptions' which
  couldn't be found
Call Stack (most recent call first):
  /home/mlanting/ros2_moveit_ws/install/share/moveit_core/cmake/moveit_coreConfig.cmake:38 (include)
  CMakeLists.txt:31 (find_package)

---
Failed   <<< moveit_ros_perception	[ Exited with code 1 ]
~~~~

full output here:
[moveit_build_error_output.txt](https://github.com/AcutronicRobotics/moveit2/files/3258644/moveit_build_error_output.txt)

I played around in the CMakeLists file for moveit_core and shifted the order libraries are listed in the libraries set, and the error seems to specifiy whichever library is first in the list.

LanderU, 2019-06-06:
Did you've downloaded the external dependencies using `vcs import src < src/moveit2/moveit2.repos` on you workspace?

LanderU, 2019-06-06:
Sorry I didn't read well, I'll have a look, thanks for reporting!

anasarrak, 2019-06-06:
Thanks for reporting @mlanting, but I'm not able to reproduce your problem. Have you tried just by compling with ``colcon build --merge-install --cmake-args -DBUILD_TESTING=FALSE`` without the _--symlink-install_?

mlanting, 2019-06-06:
@anasarrak Yes, I see the same error without --symlink-install

pjreed, 2019-06-06:
I see the same message, but mine is just a `CMake Warning` rather than a `CMake Error`, and so this build passes:
```
Starting >>> moveit_ros_perception
--- stderr: moveit_ros_perception                                         
CMake Warning at /home/preed/src/moveit/install/share/moveit_core/cmake/ament_cmake_export_libraries-extras.cmake:116 (message):
  Package 'moveit_core' exports library 'moveit_test_utils' which couldn't be
  found
Call Stack (most recent call first):
  /home/preed/src/moveit/install/share/moveit_core/cmake/moveit_coreConfig.cmake:38 (include)
  CMakeLists.txt:31 (find_package)

---
Finished <<< moveit_ros_perception [15.6s]
```

Is there perhaps something in the environment that could be causing CMake to treat that warning as an error?

For reference, my environment is:

Ubuntu: 18.04.2
cmake: 3.10.2-1ubuntu2
python3-colcon-ros: 0.3.10-1
python3-colcon-cmake: 0.2.11-1

And my `moveit2` repo is on commit `fb0e4951f70ef126547ca7b4882e78223d5c25c6`.

LanderU, 2019-06-06:
I'm trying to reproduce the issue. I'm using a docker container and the latest master version and works without problems.

Commit: fb0e4951f70ef126547ca7b4882e78223d5c25c6

mlanting, 2019-06-11:
After working with @pjreed on this, the issue seems to be related to how some of the dependencies are installed. Trying from a fresh Docker image, and using a binary install of Dashing (ros-dashing-desktop) instead of a source install, there were unmet dependencies.

On my machine I had initially started trying to get the Mara example [here](https://github.com/AcutronicRobotics/MARA#install) to run, but ran into issues and decided to simplify things by installing moveit2 on its own. That means I had installed the packages listed in the dependency list from the MARA instructions which were not installed in the container from `ros-dashing-desktop`.

I used `sudo apt install ros-dashing-gazebo-dev ros-dashing-gazebo-msgs ros-dashing-gazebo-plugins ros-dashing-gazebo-ros ros-dashing-gazebo-ros-pkgs` and no longer got the unmet dependency errors, but I did get the `Package 'moveit_core' exports...` error as before on my machine.

Then @pjreed suggested using rosdep install instead of using apt, and when I did it that way, it built successfully for me and the moveit_core export error was just a warning, as he said he was seeing when he tried replicating it.

So our best guess is that there is some dependency that causes the warning to turn into an error if said dependency is installed using apt instead of rosdep...

LanderU, 2019-06-12:
Interesting, we've problems with packages as well depending on which installation we've used. We definitively need to have a look into it. In order to build the image in our CI we are using the oficial docker image: https://github.com/AcutronicRobotics/moveit2/blob/master/.docker/ci/Dockerfile#L1 and then install the dependencies using rosdep.

mlanting, 2019-06-13:
I spent yesterday looking into this to see if I could identify which library/libraries were causing the issue, and now i'm not so sure it has to do with library installation after all...

I don't really have concrete leads since every time I think i've found a predictable pattern of behavior, I run into a counter-example, but my best guess at this point is that @anasarrak was on the right track with the --symlink-install thing. It seems more likely to break the install when --symlink-install is included, but it also breaks any future attempts to build with or without --symlink-install (which is why when I tested it without --symlink-install, it still saw the incorrect behavior).

That's not much to go on, but I don't want to leave you running down a rabbit hole with the library installation hypothesis since that pattern didn't consistently hold throughout my testing yesterday.

mlanting, 2019-06-19:
I don't know cmake very well yet, but I've tracked the error to this file where the error message actually comes from: https://github.com/ament/ament_cmake/commit/cd1236ce3de298b3bd1b7d1dd5aa4c3b376d9944#diff-29cbd162a3d3eda7d0757592dc624333

For some reason the --symlink-install, in this case, causes it to go down a different branch than without --symlink-install, and at some point the --symlink-install branch was escalated from a warning to an error.