AcutronicRobotics/moveit2 #107: OS X, control_msgs fails for OpenSplice

vmayoral, 2019-06-08:
In file included from /Users/victor/ros2_moveit_ws/build/control_msgs/rosidl_typesupport_opensplice_cpp/control_msgs/action/dds_opensplice/gripper_command__type_support.cpp:19:
In file included from /Users/victor/ros2_moveit_ws/build/control_msgs/rosidl_typesupport_opensplice_cpp/control_msgs/action/gripper_command__rosidl_typesupport_opensplice_cpp.hpp:10:
In file included from /Users/victor/ros2_moveit_ws/build/control_msgs/rosidl_typesupport_opensplice_cpp/control_msgs/action/dds_opensplice/ccpp_GripperCommand_.h:4:
/Users/victor/ros2_moveit_ws/build/control_msgs/rosidl_typesupport_opensplice_cpp/control_msgs/action/dds_opensplice/GripperCommand_.h:41:16: error: no member named 'dds_' in namespace 'control_msgs::msg'; did you mean simply 'dds_'?
               ::control_msgs::msg::dds_::GripperCommand_ command_;
               ^~~~~~~~~~~~~~~~~~~~~~~~~
               dds_
/Users/victor/ros2_moveit_ws/build/control_msgs/rosidl_typesupport_opensplice_cpp/control_msgs/action/dds_opensplice/GripperCommand_.h:24:17: note: 'dds_' declared here
      namespace dds_
                ^
/Users/victor/ros2_moveit_ws/build/control_msgs/rosidl_typesupport_opensplice_cpp/control_msgs/action/dds_opensplice/GripperCommand_.h:41:37: error: no type named 'GripperCommand_' in namespace 'control_msgs::action::dds_'; did you mean 'GripperCommand'?
               ::control_msgs::msg::dds_::GripperCommand_ command_;
                                    ^~~~~~~~~~~~~~~~~~~~~
                                    GripperCommand
/Users/victor/ros2_moveit_ws/build/control_msgs/rosidl_generator_cpp/control_msgs/action/gripper_command__struct.hpp:1217:31: note: 'GripperCommand' declared here
typedef struct GripperCommand GripperCommand;
                              ^
In file included from /Users/victor/ros2_moveit_ws/build/control_msgs/rosidl_typesupport_opensplice_cpp/control_msgs/action/dds_opensplice/joint_trajectory__type_support.cpp:19:
In file included from /Users/victor/ros2_moveit_ws/build/control_msgs/rosidl_typesupport_opensplice_cpp/control_msgs/action/joint_trajectory__rosidl_typesupport_opensplice_cpp.hpp:10:
In file included from /Users/victor/ros2_moveit_ws/build/control_msgs/rosidl_typesupport_opensplice_cpp/control_msgs/action/dds_opensplice/ccpp_JointTrajectory_.h:4:
/Users/victor/ros2_moveit_ws/build/control_msgs/rosidl_typesupport_opensplice_cpp/control_msgs/action/dds_opensplice/JointTrajectory_.h:41:16: error: no member named 'dds_' in namespace 'trajectory_msgs::msg'; did you mean simply 'dds_'?
               ::trajectory_msgs::msg::dds_::JointTrajectory_ trajectory_;
               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
               dds_
/Users/victor/ros2_moveit_ws/build/control_msgs/rosidl_typesupport_opensplice_cpp/control_msgs/action/dds_opensplice/JointTrajectory_.h:24:17: note: 'dds_' declared here
      namespace dds_
                ^
/Users/victor/ros2_moveit_ws/build/control_msgs/rosidl_typesupport_opensplice_cpp/control_msgs/action/dds_opensplice/JointTrajectory_.h:41:40: error: no type named 'JointTrajectory_' in namespace 'control_msgs::action::dds_'; did you mean 'JointTrajectory'?
               ::trajectory_msgs::msg::dds_::JointTrajectory_ trajectory_;
                                       ^~~~~~~~~~~~~~~~~~~~~~
                                       JointTrajectory
/Users/victor/ros2_moveit_ws/build/control_msgs/rosidl_generator_cpp/control_msgs/action/joint_trajectory__struct.hpp:1133:32: note: 'JointTrajectory' declared here
typedef struct JointTrajectory JointTrajectory;
                               ^
/Users/victor/ros2_moveit_ws/build/control_msgs/rosidl_typesupport_opensplice_cpp/control_msgs/action/dds_opensplice/gripper_command__type_support.cpp:2376:56: error: no viable conversion from 'Sample<control_msgs::action::dds_::GripperCommand_SendGoal_Request_>' to 'const control_msgs::action::dds_::Sample_GripperCommand_SendGoal_Request_'
    DDS::ReturnCode_t status = typed_datawriter->write(sample, DDS::HANDLE_NIL);
                                                       ^~~~~~
/Users/victor/ros2_moveit_ws/build/control_msgs/rosidl_typesupport_opensplice_cpp/control_msgs/action/dds_opensplice/GripperCommand_.h:92:17: note: candidate constructor (the implicit copy constructor) not viable: no known conversion from 'Sample<control_msgs::action::dds_::GripperCommand_SendGoal_Request_>' to 'const control_msgs::action::dds_::Sample_GripperCommand_SendGoal_Request_ &' for 1st argument
         struct Sample_GripperCommand_SendGoal_Request_
                ^
/Users/victor/ros2_moveit_ws/build/control_msgs/rosidl_typesupport_opensplice_cpp/control_msgs/action/dds_opensplice/GripperCommand_.h:92:17: note: candidate constructor (the implicit move constructor) not viable: no known conversion from 'Sample<control_msgs::action::dds_::GripperCommand_SendGoal_Request_>' to 'control_msgs::action::dds_::Sample_GripperCommand_SendGoal_Request_ &&' for 1st argument
/Users/victor/ros2_moveit_ws/build/control_msgs/rosidl_typesupport_opensplice_cpp/control_msgs/action/dds_opensplice/GripperCommand_Dcps.h:862:85: note: passing argument to parameter 'instance_data' here
            virtual DDS::Long write (const Sample_GripperCommand_SendGoal_Request_& instance_data, DDS::LongLong handle) = 0;
                                                                                    ^
/Users/victor/ros2_moveit_ws/build/control_msgs/rosidl_typesupport_opensplice_cpp/control_msgs/action/dds_opensplice/joint_trajectory__type_support.cpp:2346:56: error: no viable conversion from 'Sample<control_msgs::action::dds_::JointTrajectory_SendGoal_Request_>' to 'const control_msgs::action::dds_::Sample_JointTrajectory_SendGoal_Request_'
    DDS::ReturnCode_t status = typed_datawriter->write(sample, DDS::HANDLE_NIL);
                                                       ^~~~~~
/Users/victor/ros2_moveit_ws/build/control_msgs/rosidl_typesupport_opensplice_cpp/control_msgs/action/dds_opensplice/JointTrajectory_.h:92:17: note: candidate constructor (the implicit copy constructor) not viable: no known conversion from 'Sample<control_msgs::action::dds_::JointTrajectory_SendGoal_Request_>' to 'const control_msgs::action::dds_::Sample_JointTrajectory_SendGoal_Request_ &' for 1st argument
         struct Sample_JointTrajectory_SendGoal_Request_
                ^
/Users/victor/ros2_moveit_ws/build/control_msgs/rosidl_typesupport_opensplice_cpp/control_msgs/action/dds_opensplice/JointTrajectory_.h:92:17: note: candidate constructor (the implicit move constructor) not viable: no known conversion from 'Sample<control_msgs::action::dds_::JointTrajectory_SendGoal_Request_>' to 'control_msgs::action::dds_::Sample_JointTrajectory_SendGoal_Request_ &&' for 1st argument
/Users/victor/ros2_moveit_ws/build/control_msgs/rosidl_typesupport_opensplice_cpp/control_msgs/action/dds_opensplice/JointTrajectory_Dcps.h:863:86: note: passing argument to parameter 'instance_data' here
            virtual DDS::Long write (const Sample_JointTrajectory_SendGoal_Request_& instance_data, DDS::LongLong handle) = 0;
                                                                                     ^
3 errors generated.
3 errors generated.
make[2]: *** [CMakeFiles/control_msgs__rosidl_typesupport_opensplice_cpp.dir/rosidl_typesupport_opensplice_cpp/control_msgs/action/dds_opensplice/gripper_command__type_support.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/control_msgs__rosidl_typesupport_opensplice_cpp.dir/rosidl_typesupport_opensplice_cpp/control_msgs/action/dds_opensplice/joint_trajectory__type_support.cpp.o] Error 1
make[1]: *** [CMakeFiles/control_msgs__rosidl_typesupport_opensplice_cpp.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< control_msgs	[ Exited with code 2 ]
Aborted  <<< octomap_msgs
Aborted  <<< object_recognition_msgs

Summary: 9 packages finished [14.2s]
  1 package failed: control_msgs
  2 packages aborted: object_recognition_msgs octomap_msgs
  1 package had stderr output: control_msgs
  10 packages not processed
The file /Users/victor/ros2_moveit_ws/None does not exist.

vmayoral, 2019-06-08:
Note that I'm sourcing `release.com` as indicated at https://index.ros.org/doc/ros2/Installation/Crystal/OSX-Install-Binary/#enable-opensplice-support

ahcorde, 2019-06-08:
This Issue is reported here  https://github.com/ADLINK-IST/opensplice/issues/92

vmayoral, 2019-06-08:
> This Issue is reported here ADLINK-IST/opensplice#92

Thanks a lot!