AcutronicRobotics/moveit2 #111: Issues building tf2_geometry_msgs

vmayoral, 2019-06-11:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: Dashing
* OS Version: Ubuntu 18.04.2 LTS
* Source build

### Steps to reproduce
When building the moveit2 ws from source:

```bash
Starting >>> tf2_geometry_msgs
--- stderr: tf2_geometry_msgs
CMakeFiles/test_tf2_geometry_msgs.dir/test/test_tf2_geometry_msgs.cpp.o: In function `main':
test_tf2_geometry_msgs.cpp:(.text+0x2ab4): undefined reference to `tf2_ros::Buffer::Buffer(std::shared_ptr<rclcpp::Clock>, std::chrono::duration<long, std::ratio<1l, 1000000000l> >)'
collect2: error: ld returned 1 exit status
make[2]: *** [test_tf2_geometry_msgs] Error 1
make[1]: *** [CMakeFiles/test_tf2_geometry_msgs.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< tf2_geometry_msgs	[ Exited with code 2 ]
```

Is this due to the fact that Dashing ROS 2 installation has issues with `geometry2`?

LanderU, 2019-06-12:
We need to update the instructions. On GNU/Linux we only need to compile tf2_kdl the rest of the packages are already installed via `apt` or `rosdep`.

vmayoral, 2019-06-12:
Related to https://github.com/ros2/geometry2/issues/116. A fix is currently being applied in the CI (credit to @LanderU) https://github.com/AcutronicRobotics/moveit_ci/blob/dashing/travis.sh#L174-L175 that fetches this package directly from the sources and compiles it.

LanderU, 2019-06-15:
https://github.com/AcutronicRobotics/moveit2/pull/116

vmayoral, 2019-06-15:
Shall we close this then?

closed 2019-06-15 by LanderU