AcutronicRobotics/moveit2 #112: Issues building moveit2 with --

vmayoral, 2019-06-12:
### Description

There seems to be a problem with the existing codebase of moveit2 when built with `--symlink-install` (fails to build) versus when built with `--merge-install`.

### Your environment
* ROS Distro: Dashing
* OS Version: OS X 10.14 (validated as well with Ubuntu 18.04)
* Source

### Steps to reproduce
#### `--merge-install`
```bash
colcon build --merge-install --cmake-args -DOSRF_TESTING_TOOLS_CPP_DISABLE_MEMORY_TOOLS=ON -DINSTALL_EXAMPLES=OFF -DSECURITY=ON --no-warn-unused-cli -DCMAKE_BUILD_TYPE=Debug
/usr/local/lib/python3.7/site-packages/colcon_core/executor/__init__.py:323: UserWarning: The ExecutorExtensionPoint 'parallel' uses a deprecated signature for the 'execute' method
  .format_map(locals()))
Starting >>> eigen_stl_containers
Starting >>> joint_state_publisher
Starting >>> object_recognition_msgs
Starting >>> octomap_msgs
Starting >>> random_numbers
Starting >>> urdfdom_py
...
/Users/victor/ros2_moveit_ws/src/moveit2/moveit_core/collision_distance_field/include/moveit/collision_distance_field/collision_world_distance_field.h:151:93: warning: unused parameter 'world' [-Wunused-parameter]
  void distanceWorld(const DistanceRequest& req, DistanceResult& res, const CollisionWorld& world) const override
                                                                                            ^
98 warnings generated.
---
Finished <<< moveit_core [2min 30s]
Starting >>> moveit_ros_perception
```

#### `--symlink-install`
```bash
colcon build --symlink-install --cmake-args -DOSRF_TESTING_TOOLS_CPP_DISABLE_MEMORY_TOOLS=ON                                  -DINSTALL_EXAMPLES=OFF -DSECURITY=ON --no-warn-unused-cli                                  -DCMAKE_BUILD_TYPE=Debug
/usr/local/lib/python3.7/site-packages/colcon_core/executor/__init__.py:323: UserWarning: The ExecutorExtensionPoint 'parallel' uses a deprecated signature for the 'execute' method
  .format_map(locals()))
Starting >>> eigen_stl_containers
Starting >>> joint_state_publisher
Starting >>> object_recognition_msgs
Starting >>> octomap_msgs
Starting >>> random_numbers
Starting >>> urdfdom_py
...
In file included from /Users/victor/ros2_moveit_ws/src/moveit2/moveit_core/collision_detection/include/moveit/collision_detection/collision_common.h:47:
/Users/victor/ros2_moveit_ws/src/moveit2/moveit_core/robot_model/include/moveit/robot_model/robot_model.h:43:10: fatal error: 'srdfdom/model.h' file not found
#include <srdfdom/model.h>
         ^~~~~~~~~~~~~~~~~
1 error generated.
In file included from /Users/victor/ros2_moveit_ws/src/moveit2/moveit_core/kinematics_base/src/kinematics_base.cpp:37:
/Users/victor/ros2_moveit_ws/src/moveit2/moveit_core/kinematics_base/include/moveit/kinematics_base/kinematics_base.h:299:52: warning: unused parameter 'context_state' [-Wunused-parameter]
                   const moveit::core::RobotState* context_state = NULL) const
                                                   ^
make[2]: *** [collision_detection/CMakeFiles/moveit_collision_detection.dir/src/allvalid/collision_robot_allvalid.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /Users/victor/ros2_moveit_ws/src/moveit2/moveit_core/kinematics_base/src/kinematics_base.cpp:38:
In file included from /Users/victor/ros2_moveit_ws/src/moveit2/moveit_core/robot_model/include/moveit/robot_model/joint_model_group.h:41:
/Users/victor/ros2_moveit_ws/src/moveit2/moveit_core/robot_model/include/moveit/robot_model/joint_model.h:46:10: fatal error: 'random_numbers/random_numbers.h' file not found
#include <random_numbers/random_numbers.h>
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /Users/victor/ros2_moveit_ws/src/moveit2/moveit_core/collision_detection/src/allvalid/collision_world_allvalid.cpp:37:
In file included from /Users/victor/ros2_moveit_ws/src/moveit2/moveit_core/collision_detection/include/moveit/collision_detection/allvalid/collision_world_allvalid.h:40:
In file included from /Users/victor/ros2_moveit_ws/src/moveit2/moveit_core/collision_detection/include/moveit/collision_detection/collision_world.h:42:
In file included from /Users/victor/ros2_moveit_ws/src/moveit2/moveit_core/collision_detection/include/moveit/collision_detection/collision_matrix.h:40:
In file included from /Users/victor/ros2_moveit_ws/src/moveit2/moveit_core/collision_detection/include/moveit/collision_detection/collision_common.h:47:
/Users/victor/ros2_moveit_ws/src/moveit2/moveit_core/robot_model/include/moveit/robot_model/robot_model.h:43:10: fatal error: 'srdfdom/model.h' file not found
#include <srdfdom/model.h>
         ^~~~~~~~~~~~~~~~~
/Users/victor/ros2_moveit_ws/src/moveit2/moveit_core/profiler/src/profiler.cpp:177:32: warning: data argument not used by format string [-Wformat-extra-args]
  RCUTILS_LOG_INFO("profiler", ss.str().c_str());
                   ~~~~~~~~~~  ^
/Users/victor/ros2_ws/install/include/rcutils/logging_macros.h:492:5: note: expanded from macro 'RCUTILS_LOG_INFO'
    __VA_ARGS__)
    ^~~~~~~~~~~
/Users/victor/ros2_ws/install/include/rcutils/logging_macros.h:72:64: note: expanded from macro 'RCUTILS_LOG_COND_NAMED'
      rcutils_log(&__rcutils_logging_location, severity, name, __VA_ARGS__); \
                                                               ^~~~~~~~~~~
1 warning and 1 error generated.
make[2]: *** [kinematics_base/CMakeFiles/moveit_kinematics_base.dir/src/kinematics_base.cpp.o] Error 1
make[1]: *** [kinematics_base/CMakeFiles/moveit_kinematics_base.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
In file included from /Users/victor/ros2_moveit_ws/src/moveit2/moveit_core/utils/src/robot_model_test_utils.cpp:41:
/Users/victor/ros2_moveit_ws/src/moveit2/moveit_core/utils/include/moveit/utils/robot_model_test_utils.h:44:10: fatal error: 'srdfdom/srdf_writer.h' file not found
#include <srdfdom/srdf_writer.h>
         ^~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
make[2]: *** [collision_detection/CMakeFiles/moveit_collision_detection.dir/src/allvalid/collision_world_allvalid.cpp.o] Error 1
make[1]: *** [collision_detection/CMakeFiles/moveit_collision_detection.dir/all] Error 2
In file included from /Users/victor/ros2_moveit_ws/src/moveit2/moveit_core/planning_request_adapter/src/planning_request_adapter.cpp:37:
In file included from /Users/victor/ros2_moveit_ws/src/moveit2/moveit_core/planning_request_adapter/include/moveit/planning_request_adapter/planning_request_adapter.h:41:
In file included from /Users/victor/ros2_moveit_ws/src/moveit2/moveit_core/planning_interface/include/moveit/planning_interface/planning_interface.h:42:
In file included from /Users/victor/ros2_moveit_ws/src/moveit2/moveit_core/planning_interface/include/moveit/planning_interface/planning_response.h:40:
In file included from /Users/victor/ros2_moveit_ws/src/moveit2/moveit_core/robot_trajectory/include/moveit/robot_trajectory/robot_trajectory.h:42:
In file included from /Users/victor/ros2_moveit_ws/src/moveit2/moveit_core/robot_state/include/moveit/robot_state/robot_state.h:41:
/Users/victor/ros2_moveit_ws/src/moveit2/moveit_core/robot_model/include/moveit/robot_model/robot_model.h:43:10: fatal error: 'srdfdom/model.h' file not found
#include <srdfdom/model.h>
         ^~~~~~~~~~~~~~~~~
1 error generated.
make[2]: *** [utils/CMakeFiles/moveit_test_utils.dir/src/robot_model_test_utils.cpp.o] Error 1
make[1]: *** [utils/CMakeFiles/moveit_test_utils.dir/all] Error 2
1 warning generated.
1 error generated.
make[2]: *** [planning_request_adapter/CMakeFiles/moveit_planning_request_adapter.dir/src/planning_request_adapter.cpp.o] Error 1
make[1]: *** [planning_request_adapter/CMakeFiles/moveit_planning_request_adapter.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< moveit_core	[ Exited with code 2 ]
```

mlanting, 2019-06-13:
This might be related to what I was running into with issue [#104](https://github.com/AcutronicRobotics/moveit2/issues/104). Granted, the errors I've been seeing were different, but I had been using --merge-install and --symlink-install at the same time.

kazuki0824, 2019-07-05:
I think it's probably fixed perfect in #125 .

anasarrak, 2019-07-08:
> I think it's probably been fixed perfect in #125 .

There are some issues on my case, I'll try to fix them

```bash
--- stderr: moveit_core                                
/mnt/myfiles/github/moveitest/src/moveit2/moveit_core/version/version.cpp: In function ‘int main(int, char**)’:
/mnt/myfiles/github/moveitest/src/moveit2/moveit_core/version/version.cpp:40:14: warning: unused parameter ‘argc’ [-Wunused-parameter]
 int main(int argc, char** argv)
              ^~~~
/mnt/myfiles/github/moveitest/src/moveit2/moveit_core/version/version.cpp:40:27: warning: unused parameter ‘argv’ [-Wunused-parameter]
 int main(int argc, char** argv)
                           ^~~~
In file included from /mnt/myfiles/github/moveitest/src/moveit2/moveit_core/collision_detection/include/moveit/collision_detection/collision_common.h:47:0,
                 from /mnt/myfiles/github/moveitest/src/moveit2/moveit_core/collision_detection/include/moveit/collision_detection/collision_matrix.h:40,
                 from /mnt/myfiles/github/moveitest/src/moveit2/moveit_core/collision_detection/include/moveit/collision_detection/collision_robot.h:40,
                 from /mnt/myfiles/github/moveitest/src/moveit2/moveit_core/collision_detection/include/moveit/collision_detection/allvalid/collision_robot_allvalid.h:40,
                 from /mnt/myfiles/github/moveitest/src/moveit2/moveit_core/collision_detection/src/allvalid/collision_robot_allvalid.cpp:37:
/mnt/myfiles/github/moveitest/src/moveit2/moveit_core/robot_model/include/moveit/robot_model/robot_model.h:43:10: fatal error: srdfdom/model.h: No such file or directory
 #include <srdfdom/model.h>
          ^~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [collision_detection/CMakeFiles/moveit_collision_detection.dir/src/allvalid/collision_robot_allvalid.cpp.o] Error 1
make[1]: *** [collision_detection/CMakeFiles/moveit_collision_detection.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
In file included from /mnt/myfiles/github/moveitest/src/moveit2/moveit_core/robot_model/include/moveit/robot_model/joint_model_group.h:41:0,
                 from /mnt/myfiles/github/moveitest/src/moveit2/moveit_core/kinematics_base/src/kinematics_base.cpp:38:
/mnt/myfiles/github/moveitest/src/moveit2/moveit_core/robot_model/include/moveit/robot_model/joint_model.h:46:10: fatal error: random_numbers/random_numbers.h: No such file or directory
 #include <random_numbers/random_numbers.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [kinematics_base/CMakeFiles/moveit_kinematics_base.dir/src/kinematics_base.cpp.o] Error 1
make[1]: *** [kinematics_base/CMakeFiles/moveit_kinematics_base.dir/all] Error 2
In file included from /mnt/myfiles/github/moveitest/src/moveit2/moveit_core/utils/src/robot_model_test_utils.cpp:41:0:
/mnt/myfiles/github/moveitest/src/moveit2/moveit_core/utils/include/moveit/utils/robot_model_test_utils.h:44:10: fatal error: srdfdom/srdf_writer.h: No such file or directory
 #include <srdfdom/srdf_writer.h>
          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [utils/CMakeFiles/moveit_test_utils.dir/src/robot_model_test_utils.cpp.o] Error 1
make[1]: *** [utils/CMakeFiles/moveit_test_utils.dir/all] Error 2
In file included from /mnt/myfiles/github/moveitest/src/moveit2/moveit_core/profiler/include/moveit/profiler/profiler.h:63:0,
                 from /mnt/myfiles/github/moveitest/src/moveit2/moveit_core/profiler/src/profiler.cpp:37:
/mnt/myfiles/github/moveitest/src/moveit2/moveit_core/profiler/src/profiler.cpp: In member function ‘void moveit::tools::Profiler::console()’:
/opt/ros/dashing/include/rcutils/logging_macros.h:72:75: warning: too many arguments for format [-Wformat-extra-args]
       rcutils_log(&__rcutils_logging_location, severity, name, __VA_ARGS__); \
                                                                           ^
/opt/ros/dashing/include/rcutils/logging_macros.h:489:3: note: in expansion of macro ‘RCUTILS_LOG_COND_NAMED’
   RCUTILS_LOG_COND_NAMED( \
   ^~~~~~~~~~~~~~~~~~~~~~
/mnt/myfiles/github/moveitest/src/moveit2/moveit_core/profiler/src/profiler.cpp:177:3: note: in expansion of macro ‘RCUTILS_LOG_INFO’
   RCUTILS_LOG_INFO("profiler", ss.str().c_str());
   ^
make: *** [all] Error 2
---
Failed   <<< moveit_core	[ Exited with code 2 ]

Summary: 11 packages finished [17.1s]
  1 package failed: moveit_core
  1 package had stderr output: moveit_core
  8 packages not processed
```

anasarrak, 2019-07-08:
For some reason moveit_core is not exporting correctly the libraries

```bash
Starting >>> moveit_ros_perception
--- stderr: moveit_ros_perception                                 
CMake Error at /mnt/myfiles/github/moveitlast/install/moveit_core/share/moveit_core/cmake/ament_cmake_export_libraries-extras.cmake:48 (message):
  Package 'moveit_core' exports the library 'moveit_exceptions' which
  couldn't be found
Call Stack (most recent call first):
  /mnt/myfiles/github/moveitlast/install/moveit_core/share/moveit_core/cmake/moveit_coreConfig.cmake:38 (include)
  CMakeLists.txt:31 (find_package)

---
Failed   <<< moveit_ros_perception	[ Exited with code 1 ]

Summary: 12 packages finished [2min 19s]

```

anasarrak, 2019-07-08:
With **--merge-install** compiles without problems

> For some reason moveit_core is not exporting correctly the libraries
> 
> ```shell
> Starting >>> moveit_ros_perception
> --- stderr: moveit_ros_perception                                 
> CMake Error at /mnt/myfiles/github/moveitlast/install/moveit_core/share/moveit_core/cmake/ament_cmake_export_libraries-extras.cmake:48 (message):
>   Package 'moveit_core' exports the library 'moveit_exceptions' which
>   couldn't be found
> Call Stack (most recent call first):
>   /mnt/myfiles/github/moveitlast/install/moveit_core/share/moveit_core/cmake/moveit_coreConfig.cmake:38 (include)
>   CMakeLists.txt:31 (find_package)
> 
> 
> ---
> Failed   <<< moveit_ros_perception	[ Exited with code 1 ]
> 
> Summary: 12 packages finished [2min 19s]
> ```